bambulab/BambuStudio #4747: Bambu Studios Scaling - WIN10

RedRobin08, 2024-09-03:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

WIN10

### Printer

P1S

### How to reproduce

Load an file into Bambu Studios.  Select file.  Click to "Scale" object.  turn OFF "uniform scaling".  Attempt to Scale Z axis only

### Actual results

When I try to scale only the Z axis, the X and/or Y axis change and one dimensions appears flat (like a piece of paper).  no amount of trying to scale the object back will fix it (other than clicking undo).  The item looks unrecognizable.  

### Expected results

The item should scale only the Z axis

### Project file & Debug log uploads

[SubmitToBambu.zip](https://github.com/user-attachments/files/16842583/SubmitToBambu.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-09-03:
[RedRobin08](https://github.com/RedRobin08)：
Could you please provide a video or a gif，we will check it

RedRobin08, 2024-09-03:
This happens on all files if I try to scale with uniform scaling turned off

Haidiye00, 2024-09-04:
The issue has been fixed and the function is being tested internally. Please wait~

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz