bambulab/BambuStudio #4751: Fuzzy Skin does not work with reduced speed on overhangs

fridolin-parax, 2024-09-03:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

ryzen 7

### Printer

Bambu X1 

### How to reproduce

Load a Model with overhang, select the option use reduced speed for overhang and use the option fuzzy skin. When fuzzy skin is activated and you check the speed in the sliced preview you'll see that there is no speed reduction in the areas of overhang. It seems the fuzzy skin option is overwriting the speed setting of reduced speed.
Without fuzzy skin:
[image]
With fuzzy skin:
[image]

### Actual results

Overhangs are printed really badly with fuzzy skin

### Expected results

Overhangs should print just as good as without fuzzy skin. 
This bug makes it impossible to produce properly with fuzzy skin (which is mainly usefull for production). 
A fast fix would be really appritiated.

### Project file & Debug log uploads

[Test for fuzzy Skin + Reduced speed.zip](https://github.com/user-attachments/files/16851972/Test.for.fuzzy.Skin.%2B.Reduced.speed.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-10-12:
yes. fuzzy skin does not support overhang detection. i will mark this as a feature.

fridolin-parax, 2024-10-14:
Thanks!
Do you now how long this will take?

Best

Fridolin

_______________________________________________________________
Fridolin Förster
CTO / Technischer Leiter
*#gerneperdu*

PARAX GmbH
Bockeldamm 19
59199 Bönen
D – Germany

Kontakt:
Mobil:      +49 (0) 163 4262 787
E-Mail:     ***@***.***
Web:        *www.parax.de* <www.parax.de>

QingZhangBambu, 2024-10-14:
I am not sure, concerned staff will review the feature. may take a long time

M-A-B007, 2025-05-13:
+1
It's still an annoying issue in version 2.0.3.54.
Is there any workaround to reduce the speed in the specific areas??