bambulab/BambuStudio #4754: Editing the price of a filament is not possible on a system preset. This results in incorrect costing of prints 100% of the time

dhodgin, 2024-09-04:
### Bambu Studio Version

1.9.4.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 pro

### Additional system information

Not applicable.

### Printer

Bambu Lab X1C

### How to reproduce

1. Goto the Filament section of the left side panel
2. Click on the "Click to edit preset" edit button next to any filament
3. From the Filament settings popup, with advanced slider on, under Basic Information, the "Price" box shows the price of the filament.
4. Update the price of the filament.
5. Click the "Save current filament" save button

You will get an error if you try to save over the current preset name and not make a copy of the preset.

### Actual results

[image]

You will get an error if you try to save over the current preset name and not make a copy of the preset.

### Expected results

I expect that I should be allowed to update the price of the system profile preset and save it.

The price is hardcoded in USD currency and doesn't account for if you got some on sale or if you are in another country. As a result, you cannot ever rely on the slicer telling you the cost of the print in materials. It will always be wrong if you do not live in the USA because you do not use USD as a currency.

I cannot update my price to reflect the price I paid in CAD in Canada and as a result, I cannot ever rely on the price given to me in the slicer as the cost of my materials to price a print job. It will be off by 40% at least because the exchange rate to CAD is roughly 35-40% on a given day more.

ABS for example is more like $40.00 CAD per roll. The system is hard coded to $24.99 USD as the price of a roll.

The system will let me create a new profile but it is impractical to expect users to create new preset copies of every filament type just to change the price of the roll of filament AND then also have to select that preset everytime you load a roll of filament.

The whole purpose of the RFID chips in a Bambu roll is that the AMS can read a roll of Bambu filament and setup everything automatically. If i have to now manually select a different preset to get the correct price of a print there is no reason to buy Bambu filament for the RFID benefit. I should just buy cheaper filament from Amazon if I'm going to have to manually setup the preset after anyway to correct the pricing.

### Project file & Debug log uploads

Not Applicable

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-09-05:
This requirement has been recorded. Please wait for evaluation or design of a solution and we will contact you again. Thank you.

AngraNET, 2024-12-19:
I second this even in the US. The price of PETG-HF is wrong in system presets and I can't update it without creating a copy.

normaligator, 2025-02-18:
Hey I'm having the same issue with Bambu Lab system filament, the pricing in the software is not reflecting the real price. And when I change the value in the software it wants me to save a copy profile. I'm don't want to copy the profile, only change the pricing.

PilotNubilus, 2025-07-10:
I have just had the same issue. Being able to set the price of Bambu filament would be great.