bambulab/BambuStudio #4760: Using Ironing Type: Topmost surface, fails and causes head to embed itself in layer.

spyrule, 2024-09-06:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 23H2 (22631.4037)

### Additional system information

AMD Ryzen 7 7800X3D, 32GB DDR5, NVMe OS.

### Printer

Bambulabs P1S latest firmware

### How to reproduce

Take a print that has a flat surface, turn on Ironing Type "Topmost surface", Rectilinear, 30 mm/s Speed, 8% Ironing Flow, and 0.15 Line spacing.  Wall type Arachne, default settings.

### Actual results

Same print without ironing on the left (all other settings the exact same), the failed print on the right.
[image]

### Expected results

Surface should be smooth.

### Project file & Debug log uploads

[TallShiftKnob_ABS-v8.zip](https://github.com/user-attachments/files/16909403/TallShiftKnob_ABS-v8.zip)
[debug_Fri_Sep_06_10_20_09_73824.log.zip](https://github.com/user-attachments/files/16909444/debug_Fri_Sep_06_10_20_09_73824.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

XunZhangBambu, 2024-10-11:
@spyrule 
Hello, your model has a small bottom area. When ironing is turned on, the horizontal force between the printer nozzle and the model is relatively high, making it easy to knock over the model. You can try applying some glue to the heated bed.

spyrule, 2024-10-11:
The print never failed from it coming loose. The top layer failed because the head drove itself INTO the top layer. It was still stuck to the bed when it did this.Sent from my iPhone