bambulab/BambuStudio #4761: Strange holes in imported STEP file

theosib, 2024-09-06:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Additional system information

MacBook Pro with M2 Pro

### Printer

X1C

### How to reproduce

- Create new project.
- Import STEP file (attached)
- Look at imported model

### Actual results

There is some incorrect geometry in the imported model. 
[image]

### Expected results

I also tried importing the STEP file into FreeCAD and PrusaSlicer. It's fine in both cases. Only Bambu Studio has a problem with it.
[knurl-Part.step.zip](https://github.com/user-attachments/files/16909451/knurl-Part.step.zip)

### Project file & Debug log uploads

I don't think you need a project file. I just 
[knurl-PartCut001.3mf.zip](https://github.com/user-attachments/files/16909486/knurl-PartCut001.3mf.zip)

There is no crash, and there's no log from today anyway.

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-09-10:
HI [theosib](https://github.com/theosib)：Please try the following the steps
[image]

MackBambu, 2024-10-11:
[image]
Have you tested 1.9.5.52 ? I didn't reproduce this issue.

MackBambu, 2024-10-11:
[image]
Are there some stray edges that can be deleted? This seems to be the reason for the non-manifold edge error message

MackBambu, 2024-10-11:
[image]
[image]

There are some problems with this step model.

closed 2025-04-29 by MackBambu