bambulab/BambuStudio #4762: Rotated text generates individual letters backwards or with incorrect rotation depending on text length

dagardiner, 2024-09-06:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

x1 Carbon

### How to reproduce

Scenario: creating nametags for cubicals, with names added to a "blank" stl model in Bambu Studio using the text tool, and with the faceplate touching the build plate, so the text is rotates to show properly when the nametag is removed.

When adding names, I found that I needed to add a space to the end of the bottom line of text (last name) to get the text to render properly if there was an odd number of characters, and adding an extra space in the case of an even number of characters would also trigger the issue.  In further testing, the behavior (which error would show) would change based on the position of the first name - I'm guessing related to object collisions.  If there were an odd number of characters in the last name, one of a few things would happen to the middle letter:
- It would show up backwards to the other characters
- It would be rotated 90 degrees on the printer's x axis, so it would print perpendicularly to the rest of the letters
- It would be offset 180 degrees on the 7 axis so it stuck out below the rest of the model

If I ran a slice, it would slice it as rendered, with the reversal or incorrect rotation of the single character.  I included a 3mf with just one nameplate that exhibits the issue along with a couple screenshot examples; I can include the file with 12 nametags for more examples if it would be helpful.  I included the log file, but I don't expect it to show anything of value.

### Actual results

The middle character in the last name would be:
- It would show up backwards to the other characters
- It would be rotated 90 degrees on the printer's x axis, so it would print perpendicularly to the rest of the letters
- It would be offset 180 degrees on the 7 axis so it stuck out below the rest of the model

### Expected results

The text should be rendered normally

### Project file & Debug log uploads

[Text Rendering Error.zip](https://github.com/user-attachments/files/16909729/Text.Rendering.Error.zip)
[debug_Fri_Sep_06_09_33_50_74896.log.zip](https://github.com/user-attachments/files/16909749/debug_Fri_Sep_06_09_33_50_74896.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-09-09:
Thank you for discovering the issue. Could you please record another GIF for us to analyze your operation? Thank you

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00