bambulab/BambuStudio #4764: 对于非常接近平面但不是平面的面，使用STP文件切片时可能会出现异常表面纹理

iamapig120, 2024-09-06:
### Bambu Studio Version

1.9.5

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11 22621.4037

### Printer

Bambu Lab P1S

### How to reproduce

1. 导入模型，并按照图示
[image]
角度摆放
2. 使用0.12层高和任意墙生成器切片
3. 在图示区域出现了异常的外墙走线
[image]

### Actual results

使用Step文件切片异常，出现了不该出现的走线，这个问题同样发生在BambuStudio的分支Orca上，但没有在Prusa上复现，

如果使用高精度STL文件，则可同时在Prusa和BambuStudio上复现，推测可能是早期从Prusa分支时对于墙生成器的某个修改导致了该问题。

### Expected results

切片正常，不应出现错误的表面墙走线，截图来自Prusa
[image]

### Project file & Debug log uploads

[外壳-edit.zip](https://github.com/user-attachments/files/16910758/-edit.zip)
附件包含了用于切片的原始step文件，最初使用SolidWorks导出，通过Fusion编辑后重新导出

### Checklist of files to include

- [ ] Log file
- [ ] Project file

iamapig120, 2024-09-10:
烦请催促复现情况？另，两种墙生成器都有此bug

QingZhangBambu, 2024-09-18:
@iamapig120 你好，请确认一下是否是略微旋转0.1或是0.2°就会切片正确。以及，是否关闭悬垂降速后就会切片正确。

QingZhangBambu, 2024-09-18:
@iamapig120 如果可以请上传对应的3mf，这个错误切片特征可能仅在特殊角度出现。需要3mf文件进行验证。

iamapig120, 2024-09-19:
> @iamapig120 如果可以请上传对应的3mf，这个错误切片特征可能仅在特殊角度出现。需要3mf文件进行验证。

问题已经在1.9.7中得到修复，感谢

close#4767

iamapig120, 2024-09-19:
是否关闭悬垂降速对切片没有影响
旋转会导致表面图案变化

以下是提供的3mf文件
[外壳-edit.zip](https://github.com/user-attachments/files/17066302/-edit.zip)

iamapig120, 2024-09-24:
> @iamapig120 如果可以请上传对应的3mf，这个错误切片特征可能仅在特殊角度出现。需要3mf文件进行验证。

已上传，烦请跟进？@QingZhangBambu

iamapig120, 2024-10-08:
@QingZhangBambu 3mf已上传至comment中，烦请跟进？

QingZhangBambu, 2024-10-11:
@iamapig120 好的。我看到了你的模型。会进行跟进

XunZhangBambu, 2024-10-11:
@iamapig120 
[image]
[image]

异常纹理是由圆弧拟合造成的，关闭圆弧拟合后，arachne模式下的异常纹理会减少。而使用经典墙生成器时，几乎无可见异常纹理

iamapig120, 2024-10-11:
> @iamapig120 [...]

了解了，我会尝试使用经典墙生成器打印一个测试件，这个模型顶部非常难以打印，希望在关闭圆弧拟合和使用经典墙生成器状态下能够表面不出现瑕疵。

以及还是希望能够尽量解决这个表面异常纹理的问题，使其在大多数打印配置下不出现问题。