bambulab/BambuStudio #4766: Can't add Bambu Glow filament profile to A1 Mini in Bambu Studio

logal727, 2024-09-06:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.6.1

### Additional system information

Apple M1 Max, 32GB

### Printer

A1 Mini

### How to reproduce

Select A1 Mini as printer

Go to Filament and Select Add/Remove filaments

Check Bambu Glow under System Filaments

Bambu Glow does not show up in system filaments drop down. 

### Actual results

Not able to select Bambu Glow profile when using A1 Mini

I am able to when I have my P1P selected, but not for A1 Mini

### Expected results

I should be able to use the Bambu Glow profile for A1 Mini

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16912799/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DanBao-Bambu, 2024-09-10:
Sorry for any inconvenience caused, A1 mini currently does not allow the selection of the preset file Bambu PLA Glow.

XunZhangBambu, 2024-09-10:
Due to structural reasons, when A1 mini prints this filament, there exists some friction. In order to prevent it from affecting the normal printing of other filaments, we do not allow A1 mini to print this filament.

codyrgardner, 2024-10-06:
Is the A1 impacted by the structural issues as well?

barshow, 2025-02-13:
@XunZhangBambu https://us.store.bambulab.com/products/pla-glow does not mention this.  This needs to be addressed, or at least noted on the purchase page.