bambulab/BambuStudio #4768: Print a vertical grip on the test line

TennisRunner, 2024-09-07:
**Is your feature request related to a problem? Please describe.**
Sometimes when I remove the test line that is printed at the start at the edge. It slips and stabs under my finger nail.

**Describe the solution you'd like**
At the end of the test strip, it would be nice to have it print a small vertical tab that you can pinch in order to peel it off.

**Describe alternatives you've considered**
Yes I could use a scraper, but I'm often misplacing it cus donkey brains.

**Additional context**
Example model:
[image]

lar3ry, 2024-09-13:
How would the printer handle the rest of the test line? It can' rise up then drop down without movng somewhere away from the vertical.
A better solution would be to use your fingernail to pull the left side of the line toward you. I do that all the time and have nnot stabbed myself with it yet.

TennisRunner, 2024-09-19:
> How would the printer handle the rest of the test line? It can't rise up then drop down without movng somewhere away from the vertical. A better solution would be to use your fingernail to pull the left side of the line toward you. I do that all the time and have nnot stabbed myself with it yet.

I don't understand what you mean by this. 
The vertical grip would be at the end of the test line.
I already use my nails, but some petg filaments are extra strong and require a lot of force to slip a nail under. Once it finally goes through, my finger continues move forward from inertia and the edge of the filament stabs me under the nail. 

Its not a high priority issue, just an idea worth considering.

lar3ry, 2024-09-19:
> I don't understand what you mean by this. The vertical grip would be at the end of the test line. I already use my nails, but some petg filaments are extra strong and require a lot of force to slip [...]

From the way you drew the test line, your proposed vertical pie would not be at the end. It would be in the middle.
The line starts at the left, goes right, then back, then left to finish. It makes sense only if you are proposing to  start the test line at the left, thengo right, left, then back, then up, giving you a single vertical.

On the other hand, if you remove it as I do, the problem is completely solved. What I don't understand is why you would try to remove the line with your fingernai , with your finger pointing directly at the end of a line. Even a stongly adhered line can be removed safely with a fingernail with the finger near the end of the line, and pointed across the line.

I think I learned that the hard way, probably 60 or  70 years ago.

> Its not a high priority issue, just an idea worth considering.

TennisRunner, 2024-09-19:
> From the way you drew the test line, your proposed vertical pie would not be at the end. It would be in the middle. The line starts at the left, goes right, then back, then left to finish. It makes [...]

I probably don't keep my fingernails as long as you do. Doing it with the fingerprint side down would result in my nails not reaching the bed at all. My nail/thumb would have to be perpendicular to the filament to have the nail touch the bed at all. 

The lip can be printed on the left finish side or it can go back and print it on top of the right side. I have no preference on that.