bambulab/BambuStudio #4769: Support bases placed outside the build plate for large buildplate-filling items.

onthuhlist, 2024-09-07:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

The absolute latest. Sonoma 14.6.1

### Additional system information

Mac Mini with M1 chip.

### Printer

A1 with AMS

### How to reproduce

I am printing a large 250mm cube box on the A1.. The box design has lots of holes in it that are triangular and hexagonal in shape. This hole pattern is also on the base of the box.  I printed another design of box of this size before without supports, but the current design requires supports. I prefer tree supports because they can be easily removed, and it's easy to keep their bases on the baseplate, as supports placed on the model itself are typically much harder to remove. Well, for some reason when I have the option checked to generate tree supports on build plate only, it places the bases of those tree supports both inside the box (with the tree bases being placed in the holes in the box pattern) and also outside the box. Obviously those outside the box create G-code paths that are outside the physical capabilities of the 256mmx256mm build plate. I tried adding support blockers outside the box, but support blockers seem to only apply to the physical model and WHETHER a support is generated for that model area, and not WHERE the support is actually placed. So there is nothing I can do to keep tree supports INSIDE the box only. I had to switch to dumb regular supports, printed on the model itself, so now I have dozens and dozens of little support blanks to punch out of the pattern afterwards, and I know it's going to leave a much rougher surface than if I could have used tree supports. The file I am sending you, Big Box 2, has 4 support exclusion zones added outside the box. You can remove these if you want. They do nothing to hinder generation of tree supports outside the box, even though I slightly overlapped them with the part model. To reproduce the issue I am seeing, change the supports from normal (auto) to tree (auto),  and check "on build plate only". Keep "support critical regions only" unchecked.

### Actual results

Tree supports are generated outside the 250mmx250mm box, when it is entirely possible to support all of the necessary bridge areas with tree supports within the box. By the way, to get this model to print with normal supports, I had to change the Initial layer expansion setting from 2mm to 0mm. Otherwise the normal supports would also be placed outside the build plate's capability. So this is not an issue with just tree supports.

### Expected results

I expect that when tree supports (or normal supports) are generated, the algorithm will never place them outside the build area, but will instead look for alternatives within the build area. I understand for small items that can be moved, a simple recentering of the item will fix the issue, and move the support bases so they are inside the build area. But when the item itself plus its support bases is wider than 256mm or deeper than 256mm, that is a logical impossibility. The program should then know that repositioning the item is not feasible, and should then restrain all supports within the build plate area if possible.

### Project file & Debug log uploads

[BIG BOX 2.3mf.zip](https://github.com/user-attachments/files/16914044/BIG.BOX.2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

allCanadian17, 2024-09-08:
Bambu Studio Version

1.9.5.51
Where is the application from?

Bambu Lab Official website
OS version
windows 11 professional
Additional system information

ryzen 7 3700x with rtx 3080
Printer

X1 carbon with AMS
bambu PLA silver @ 210-215 

How to reproduce 
- doent have to be this file persay but any large hlaf bowl that reaches preety close to the build plate boundries. 
- 0.2m nozzle 0.08m layher height 
- everthing is stock settings besides wall generator is set to Arachne and contour scarf joint seam 
- support settings - set as defult unless listed here - tree (auto) thresshold angle: 26 build plate only 
- initial layer expansion - 0mm 
- top z distance - 0.12mm
- interface layers - 3 
Actual results

Tree supports are generated outside the 250mmx250mm box
I am currently having the same annoying issue where I'm trying to print a large half of the stanley cup bowl and it has more than enough room to angle the tree support bases within the print bed but it keeps putting the bases of the tree supports beyond the plate making it so it is impossible to print with the "error: A g-code path goes beyond plate boundaries' always apprearing even though i change setting and make adjustment. as well as ther file fits on the plate.  

PLEASE FIX THIS MAJOR ISSUE!!!!!

Expected results
exactly what he stated above!!! 
I expect that when tree supports (or normal supports) are generated, the algorithm will never place them outside the build area, but will instead look for alternatives within the build area. I understand for small items that can be moved, a simple recentering of the item will fix the issue, and move the support bases so they are inside the build area. But when the item itself plus its support bases is wider than 256mm or deeper than 256mm, that is a logical impossibility. The program should then know that repositioning the item is not feasible, and should then restrain all supports within the build plate area if possible.

[Stanley_Bowl half.zip](https://github.com/user-attachments/files/16923331/Stanley_Bowl.half.zip)
[image]

PLEASE FIX THIS!!!!!!

tangketan, 2024-09-18:
Its'a known issue of organic support. I'll fix it in the next version.

allCanadian17, 2024-09-18:
> Its’a known issue of organic support. I'll fix it in the next version.

It is NOT just an organic support issue…. :/  or mean mean all tree supports as organic????  I tried it on every single tree support setting. I mainly was trying slim tree supports. As it is such a big print. 
I switched the file to Orca slicer and it was able to get it to slice properly with the slim supports.

I also would like to add maybe something else the program could use. I am very much a perfectionist with my prints but can the devs look into adding in the reversing directions for every other layer that orca slicer has under their quality tab it makes a big difference in the surface quality especially in the bowl part I'm trying to print 
Same with maybe the slow down for curled perimeters just a situation! :) thank you! Hopefully the fix will be out soon!!!!

tangketan, 2024-09-18:
No, not all tree supports are organic supports.
Yes, you are right. Other tree supports also have this issue. I'll fix them all.
For "reversing directions for every other layer" feature, did you mean this one?
[image]

onthuhlist, 2024-09-18:
I am not sure what you are referencing in our conversation where "reversing
directions for every other layer" may have been mentioned, as this email
does not contain the context (the conversation up to this point). That is
an important feature that tech support should consider. I currently have
aabout a half dozen or more issues open with either Bambu Studio, or
Capcut, or Macdroid which I use to transfer my video footage from my phone
to my Mac for editing. It is therefore helpful to include the context in
the email response to remind me what has been discussed up to this point.

I am not familiar with what the reversing directions setting does. I was
writing because tree support bases were being plqced both inside and
outside the walls of my 255mm x 255mm x 255mm box with holes, but the bed
can only fit the model if the tree support bases are limited to placement
inside the box, and then branch out from there to support the bridging
areas. I understand with smaller models, one would be able to move the
model so that any tree support bases could be relocated onto the print bed.
But at a certain model size, it is no longer possible to reposition the
model on the bed if the algorithm chooses to place the supports outside the
model. Granted, most models are more or less solid, in which case the
support bases MUST be placed outside the model. But in this model's case,
it would be printable with tree supports if I had an option to specify that
the tree support bases only be generated onto the print bed surface, not
outside it, when the model is correctly positioned within the pront bed
perimeter. The only way I could print this decorative box now was to use
trasitional supports in the snug setting, which resulted in several hundred
tabs to punch out, which took me hours and hours to remove because the
punchouts will not punch out when using a tough PLA which allows some
plastic deformation before the tabs break off. Many of the tabs had to be
painstakingly cut out, taking hours. But tree supports could have been
removed in 10 minutes.

fahaaaxiki, 2024-11-05:
@onthuhlist We have solved this problem in 1.10 Beta2, you can try it with https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81

closed 2026-11-03 by ArthurBambulab