bambulab/BambuStudio #4772: No geometry data

carstenpoulsen, 2024-09-07:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

X1

### How to reproduce

Load the attached 3MF file. 
[Prop25mm.zip](https://github.com/user-attachments/files/16916343/Prop25mm.zip)

### Actual results

TWO messages (why not just one????)
"Loading of model file failed" AND
"The file does not contain any geometry data" <- this message is sufficient!
[Prop25mm.zip](https://github.com/user-attachments/files/16916053/Prop25mm.zip)

The file load fine into 3D Viewer
[Prop25mm.zip](https://github.com/user-attachments/files/16915811/Prop25mm.zip)
[Prop25mm.zip](https://github.com/user-attachments/files/16916176/Prop25mm.zip)

### Expected results

Loaded image 

### Project file & Debug log uploads

[Prop25mm.zip](https://github.com/user-attachments/files/16916392/Prop25mm.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

carstenpoulsen, 2024-09-07:
[x.zip](https://github.com/user-attachments/files/16916799/x.zip)
The same image saved to a different filename works fine??????

LiZ-Li-BBL, 2024-09-24:
An parsing error occured when parsing the file [Prop25mm.zip](https://github.com/user-attachments/files/16916392/Prop25mm.zip), possibly because it was generated by a faulty older version of Bambu Studio.
You can fix this file by deleting <assemble> content in the Prop25mm.3MF/Metadata/model_settings.config file. But I recommend you just use another file that is loaded normally.

carstenpoulsen, 2024-09-24:
OK. Where do I find this Prop25mm.3MF/Metadata/model_settings.config ?

 

The faulty file is generated by SolidWorks.

But it cannot be SW that is at fault because when I save the same to a different file, it works well.

 

Br Carsten

 

You can fix this file by deleting content in the Prop25mm.3MF/Metadata/model_settings.config file. But I recommend you just use another file that is loaded normally.

LiZ-Li-BBL, 2024-09-24:
You can just open the 3mf with some decompression software.
[image]
[image]

LiZ-Li-BBL, 2024-09-24:
> You can just open the 3mf with some decompression software. [...]

delete these:
[image]

LiZ-Li-BBL, 2024-09-24:
Do you open a BambuStudio 3mf with SolidWorks? Where is the 3mf file first created?

carstenpoulsen, 2024-09-24:
My toolchain is:

Solidworks generate a 3mf file

Bambu Studio opens the 3mf

 

If I am not happy with the 3D model, I open SW and start all over.

When I create a new 3mf Bambu is not happy and declare the error I have described.

 

I can easily open the 3mf file with other tools, it is only Bambu Studio that is unable to figure out what is going on.

LiZ-Li-BBL, 2024-09-25:
I see. We will follow the steps to try to repeat the problem. I think it may take some time.

carstenpoulsen, 2024-09-25:
Good. I will on the other hand see if I can figure out a pattern 😊

carstenpoulsen, 2024-09-25:
I have just made a controlled experiment:

1.	Used SolidWorks to create a 3D model and saved it to cube.sldprt
2.	saved the model to cube.3mf <- note: lower case letters
3.	Loaded the cube.3mf file into Bambu Studio by double-clicking the 3mf file (success)
4.	Selected an AMF filament different from the default
5.	saved the project back into cube.3mf by closing Bambu Studio while affirming to save
6.	Made changes to cube.sldprt in SolidWorks
7.	saved the model to cube.3mf <- note: I was asked if I wanted to overwrite (YES)
8.	Loaded the cube.3mf file into Bambu Studio by double-clicking the 3mf file
9.	ERROR: “Loading of the model file failed”
10.	ERROR: “The file does not contain any geometric data”

Opening cube.3mf with ‘3D Viewer’ is successful.

DrRichG, 2024-10-06:
I get the same now whenever I use a  .3mf file exported from FreeCAD, when I use an .stl file instead it imports correctly

carstenpoulsen, 2024-10-06:
I think the reason is that Bambu has invented the rule that when the file extension is uppercase (.3MF) the file is interpreted as a graphics file, and when in lowercase (.3mf) the file is a Bambu Studio project file.

If you save in uppercase (default for most 3D drawing tools) as it is when using a standard file save dialog box from Windows, the extension is default uppercase. If the file is saved with an explicit extension (you type “file.3mf”) the file is saved with the typed case, the problem occurs.

Bambu must stop using the file extension case as project indicator and implement something else, eg their own file extension eg “.bambu” to ensure differenciation between a bambu studio project file and a true 3mf file.

carstenpoulsen, 2025-02-14:
Any news to issue #4772?

b1ggi, 2026-01-12:
any news? its 1 year later and still this problem.
i found out if you import the same problematic files inside the app (through File > Import > Import 3mf....) it works!

There has to be a simple solution if this is the case, right?

BTW: im using the linux version of bambustudio v2.4.0.70 as flatpak

closed 2026-09-08 by carstenpoulsen