bambulab/BambuStudio #4773: Changing "Initial layer density" in "Raft" options effects the initial layer density of supports even after disabling the raft

kumpakri, 2024-09-07:
### Bambu Studio Version

1.9.5.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Banbu lab P1S

### How to reproduce

1. Use a model with big overhangs
2. Allow Tree supports
3. In "Raft" options increase Raft layers to 1
4. Decrease Initial layer density to 40
5. Change the Raft layers back to 0
6. Slice

### Actual results

When you look at the fist layer of the supports (that are without raft now) the first layer is very sparse and would not stick to the plate. Unless you look on the fist layer in the slicer you wouldn't know, because the "Initial layer density" option is now not visible due to the raft being disabled.
[image]
[image]

### Expected results

If I don't see the setting, I expect it to have no effect on my print. I would wxpect the layer density to reset back to 90 % for the supports when the raft is disabled.
[image]

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/16917986/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-09-12:
The option is actually "support first layer density", no matter there is raft layer or not. Let me change to a proper name in the next version.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz