bambulab/BambuStudio #4775: Coordinates inside assembly are set to world coordinates instead of local coordinates.

coat64, 2024-09-07:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 5700G
64 GB
Internal graphics

### Printer

X1C

### How to reproduce

add a cube to a model, and select the cube in the assembly for moving.

### Actual results

The coordinates presented are the "world" coordinates instead of the local coordinates relative to the assembly.

### Expected results

You should see (as it was) the relative coordinates to the assembly.
Sticking to the world coordinates can be very crippling if you want to do more complicated editing within Bambu Studio. Especially aligning multiple parts is a pain now.

Can not find a setting where you can change this (new?) behavior.

### Project file & Debug log uploads

[example110.zip](https://github.com/user-attachments/files/16919852/example110.zip)
[debug_Sat_Sep_07_02_05_26_13936.log.0.zip](https://github.com/user-attachments/files/16919853/debug_Sat_Sep_07_02_05_26_13936.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

coat64, 2024-09-08:
In Orca slicer the difference between world- and object coordinates is still intact.

[image]

Haidiye00, 2024-09-09:
[image]
We have noticed this issue and the internal version has been resolved. Please stay tuned for the new version~

closed 2024-10-16 by Haidiye00