bambulab/BambuStudio #4777: Should offer a switch for user to decide instead of directly close the camer

RuioWolf, 2024-09-08:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 22h2

### Printer

X1C

### How to reproduce

commit https://github.com/bambulab/BambuStudio/commit/e3a7ad05487e9517730e8f9d7d5613ce9f28c20e introduce `stop liveview long time no input`.

### Actual results

the camera closes after 900s of idle.

### Expected results

I think it should offer a switch for user to choose whether close the camera after a certain time, instead of directly close it.

### Project file & Debug log uploads

no log needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-09-09:
the same issue #4690

ttt888zzz, 2024-10-11:
HI[KedziePanda](https://github.com/KedziePanda)、[mupke475](https://github.com/mupke475): 
1.Please download the latest version and visit the website，https://bambulab.cn/zh-cn/download/studio
2.Update 【References】setting，according the photo
[image]
https://github.com/bambulab/BambuStudio/releases

closed 2024-10-11 by ttt888zzz