bambulab/BambuStudio #4779: Variable layer height crashes on Linux

kschwi, 2024-09-08:
### Bambu Studio Version

1.9.6.51

### Where is the application from?

Bambu Lab github releases

### OS version

OpenSuSE 15.5

### Additional system information

CPU: i5-2540, 8Gb, i915 running Bambu_Studio_ubuntu-v01.09.05.51-20.04.AppImage 

### Printer

Bambu Lab P1Ps

### How to reproduce

1. Select the Buttons, 2. Click on the Menu Icon "Variable layer height" -> Studio Crahes

### Actual results

Studio crashes 

### Expected results

The dialogue to modify the layer height should open

### Project file & Debug log uploads

[arduino_uno_lcd_buttons.zip](https://github.com/user-attachments/files/16922734/arduino_uno_lcd_buttons.zip)
[debug_.zip](https://github.com/user-attachments/files/16922736/debug_.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-09-09:
Looks like the same problem.
https://github.com/bambulab/BambuStudio/issues/4417
@kschwi Can you record a video of the process repeating itself? I have not been able to reproduce this issue.

kschwi, 2024-09-09:
@MackBambu I tried the same project on a i7 with 32Gb. No problem! My guess is a problem related to a machine with small memory.
[Trace.tar.gz](https://github.com/user-attachments/files/16936656/Trace.tar.gz)
I ran Studio in gdb. The attached archive contains in trace1 all messages from the console including gdb fault detection and a stack trace. Trace2 is the same but with added debug information for the system libraries. I would propose you create a Studio containing debug information (gcc -g),  make it static to have your libs available and a dynamic one and forward it to me.
Video recording does not make sense, because starting studio, create a new project, include STL file (buttons.stl from https://www.thingiverse.com/thing:845415), selecting the buttons and clicking on "Variable layer height"-icon is all. And with the click Studio is done.

MackBambu, 2024-11-24:
> @MackBambu I tried the same project on a i7 with 32Gb. No problem! My guess is a problem related to a machine with small memory. [Trace.tar.gz](https://github.com/user-attachments/files/16936656/Trace.tar.gz) I ran Studio in gdb. The attached archive contains in trace1 all messages from the console including gdb fault detection and a stack trace. Trace2 is the same but with added debug information for the system libraries. I would propose you create a Studio containing debug information (gcc -g), make it static to have your libs available and a dynamic one and forward it to me. BTW: trace2 took ages to open the Studio window. Video recording does not make sense, because starting studio, create a new project, include STL file (buttons.stl from https://www.thingiverse.com/thing:845415), selecting the buttons and clicking on "Variable layer height"-icon is all. And with the click Studio is done.

Sorry for the delayed response. The software crash due to insufficient memory is a known issue, and it affects all features. We are considering implementing some warning messages when memory is running low.

kschwi, 2024-11-26:
@MackBambu: Your delay is no problem. I downloaded the source code and "greped" in the src-directory for malloc  and as my first guess it took me seconds to find requests for dynamic without an "if" for verification. As far as I saw there are several "malloc" and "calloc" in the tree and checking each occurrence is a lot of work. Additionally you should think about a static code analysis to find other common issues currently not detect such as usage of previously freed memory.

ladyluck9702, 2025-05-16:
I have the same issue on my brand new laptop. Everytime I click the Variable Layer Height button is just closes Bambu Studios. 

[image]
I wouldn't think this is from insufficient memory since my old computer had less memory and did not have a problem using the variable layer height feature.

ladyluck9702, 2025-05-17:
How do you find the crash logs?

ladyluck9702, 2025-05-21:
I am not finding where to get the logs. Unfortunately I am not that computer swavy.