bambulab/BambuStudio #4781: Can't Open Bambu Studio anymore

TheUnknownDan, 2024-09-08:
### Bambu Studio Version

01.09.04.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel i7-7700k, Nvidia GTX 1070, 64GB DDR4 RAM

### Printer

Bambu Lab X1C

### How to reproduce

[image]
By trying to open the program.

### Actual results

[image]

### Expected results

Program opens.

I did try to un-and reinstall the current version, did not help.

### Project file & Debug log uploads

Project file inaccesible since it wasa  crash and doesnt open anymore, but it was from the Makerworld Model https://makerworld.com/en/models/476115#profileId-387012, upscaled to 250%, split to 4 different buildplates, crash occurred after the second plate finished printing.

Log File of the Crash:
[crash_Sun_Sep_08_18_07_53_0.log](https://github.com/user-attachments/files/16923160/crash_Sun_Sep_08_18_07_53_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

TheUnknownDan, 2024-09-09:
The same issue stays even after updating to Version 01.09.05.51

lanewei120, 2024-09-10:
hello @TheUnknownDan 

could you upload the log file instead of the crash file here?
[image]

and it is very likely caused by the config file corruption
you can remove following files or directory and retry 
[image]

TheUnknownDan, 2024-09-10:
[Log Files Bambu.zip](https://github.com/user-attachments/files/16948707/Log.Files.Bambu.zip)
Since in the Folder arte 32 Log files for the day the crash happened, i'll send all of them.

also, will deleting the config file also delete my print profiles etc? 
or what will be impacted?

lanewei120, 2024-09-11:
> [Log Files Bambu.zip](https://github.com/user-attachments/files/16948707/Log.Files.Bambu.zip) Since in the Folder are 32 Log files for the day the crash happened, i'll send all of them.
> 
> also, will deleting the config file also delete my print profiles etc? or what will be impacted?

it won't delete the profiles
the profiles are located in the user directory

delete the 'system' directory has no impact;
and if delete the bambustudio.conf, the preference settings of application will be reset

lanewei120, 2024-09-11:
@TheUnknownDan the log seems abnormal, we can find limited useful informations

could you also help to do following things
1. check whether bambustudio exists in the task manager, if yes, kill them
2.  remove all the logs in the log directory
3. restart bambustudio
4. send the new log to us

thanks

TheUnknownDan, 2024-09-11:
Hi, it was not in the task manager, and the newly created log file also talks about a config file corruption.
[debug_Wed_Sep_11_20_23_16_3520.log](https://github.com/user-attachments/files/16968372/debug_Wed_Sep_11_20_23_16_3520.log)

I also noticed that i have a lot of .co0nf files, is that normal?

[image]

i've deleted the main Config file now and the program now starts correctly as if it was a completely new install,

lanewei120, 2024-09-12:
not normal
the BambuStudio.conf is corrupted

just delete them and restart software from beginning