bambulab/BambuStudio #4782: Custom profile for Qidi

pukulab, 2024-09-08:
### Bambu Studio Version

.19.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

CPU: 10 gen, i9, 64gg ram, Nvidia RTX2060

### Printer

Qidi Xmax3

### How to reproduce

custom profile for Qidi printer (printer + filamentprofile) dont save. everytime if i restart Bambustudio, the profile disappear. i ve already discuss this on forum. seems like it is a bug.

### Actual results

no solution - because it just dont save any chances

### Expected results

fix this?

### Project file & Debug log uploads

dont need :D

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-09-23:
we found an issue related with Qidi profiles

we will fix it in future versions

pukulab, 2024-09-23:
perfect - thank you very much :)

lanewei120, 2024-11-21:
already fixed in 1.10.0

closed 2024-11-21 by lanewei120