bambulab/BambuStudio #4785: Camera in Bambu Studio turning off "Temporarily closed because there is no operating for a long time."

KedziePanda, 2024-09-09:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Processor 11th Gen Intel(R) Core(TM) i9-11900KF @ 3.50GHz   3.50 GHz
Installed RAM	128 GB (128 GB usable)
Display Adapter NVIDIA GeForce RTX 3080 Ti

### Printer

Bambu Labs X1 Carbon

### How to reproduce

After updating Bambu Studio to v1.9.5.51 I've been having an issue. The live camera feed from my X1C cuts out after about 15 minutes with the message "Temporarily closed because there is no operating for a long time."

### Actual results

The camera feed in Bambu Studio times out after 15min.

### Expected results

The camera should not time out, at a minimum allow users to disable this timeout as a setting in Bambu Studio. 

From my understanding this was implemented as some complained that their computer wouldn't go to sleep. This is a rather ridiculous and inefficient solution in my opinion. Many users like myself specifically use this function to monitor prints as well as progress passively on a separate monitor and should not need to go and turn the feed back on constantly. 

If a subset of users are concerned about their PC not going to sleep, recommend they turn it off manually rather than forcing everyone else to constantly turn their feeds back on. Even better, simply make it a setting that can be chosen.

While selecting "Go Live" and letting the virtual camera adapter run in the background seems to keep the feed active it is an inefficient solution and continuously uses an extra 11w of power above and beyond the already running camera feed at idle even when the monitors are turned off/asleep, negating any energy savings of other users computers sleeping. Please revert this change or implement a way to disable this timeout.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-09-09:
The same issue https://github.com/bambulab/BambuStudio/issues/4690
#4690

mupke475, 2024-10-09:
Hallo hier hetzelfde probleem, is er misschien al een oplossing voor . Zoja graag een mailtje aub : jiboe2@hotmail.com

ttt888zzz, 2024-10-11:
HI[KedziePanda](https://github.com/KedziePanda)、[mupke475](https://github.com/mupke475): Please download the latest version and visit the website，https://bambulab.cn/zh-cn/download/studio。
[image]
https://github.com/bambulab/BambuStudio/releases

closed 2024-10-11 by ttt888zzz