bambulab/BambuStudio #4786: Camera video stops within about 5 seconds

eintr, 2024-09-09:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab github releases

### OS version

LinuxMint (Ubuntu 20.4 dirivated)

### Additional system information

Dell Precision-5820-Tower-X-Series

OS: Linux Mint 21.3 x86_64 
Host: Precision 5820 Tower X-Series 
Kernel: 5.15.0-119-generic 
Uptime: 11 days, 2 hours, 7 mins 
Packages: 2789 (dpkg) 
Shell: bash 5.1.16 
Resolution: 2560x1440 
DE: MATE 1.26.0 
WM: Metacity (Marco) 
Theme: Mint-Y-Aqua [GTK2/3] 
Icons: Mint-Y-Sand [GTK2/3] 
Terminal: mate-terminal 
Terminal Font: VictorMono Nerd Font Semi-Bold 16 
CPU: Intel i9-10900X (20) @ 4.500GHz 
GPU: NVIDIA GeForce RTX 3090 
Memory: 22489MiB / 128520MiB 

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Just Navigate to: Device -> Status
and watch the live camera.

### Actual results

Video always stops within about 5 seconds.

### Expected results

Video keeps playing.

### Project file & Debug log uploads

[debug_Mon_Sep_09_16_57_27_1289602.log.0.gz](https://github.com/user-attachments/files/16928381/debug_Mon_Sep_09_16_57_27_1289602.log.0.gz)

### Checklist of files to include

- [X] Log file
- [ ] Project file

eintr, 2024-09-09:
> Please help collect some gstreamer logs in console.

Console said nothing about gstreamer but a
`BambuTunnel::GetMsg(1)`
Are there any options to enable gstreamer logs?

I'll post another log pack with debug=5, there are some MediaPlayCtrl lines while I clicked PLAY button.
[Bambu_studio_debug.tar.gz](https://github.com/user-attachments/files/16930067/Bambu_studio_debug.tar.gz)

bambu123, 2024-09-11:
I really need all BambuTunnel log from console when playing liveview.

> From: "John ***@***.***>
> Date:  Mon, Sep 9, 2024, 20:07
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> > Please help collect some gstreamer logs in console.
> Console said nothing about gstreamer but a
> 
> BambuTunnel::GetMsg(1)
Are there any options to enable gstreamer logs?

eintr, 2024-09-11:
Here, the result of
`Bambu...... --debug 5 >console.log 2>&1`
[console.log.gz](https://github.com/user-attachments/files/16959698/console.log.gz)

bambu123, 2024-09-11:
Sorry!
Please set envriment 'GST_DEBUG=4' and run Studio from console.

> From: "John ***@***.***>
> Date:  Wed, Sep 11, 2024, 15:35
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> Here, the result of
> 
> Bambu...... --debug 5 >console.log 2>&1
> console.log.gz<https://github.com/user-attachments/files/16959698/console.log.gz>
> > I really need all BambuTunnel log from console when playing liveview.
From: "John @.> Date: Mon, Sep 9, 2024, 20:07 Subject: Re: [bambulab/BambuStudio] Camera video stops within about 5 seconds (Issue #4786<https://github.com/bambulab/BambuStudio/issues/4786>) To: @.> Cc: @.>, @.> > Please help collect some gstreamer logs in console. Console said nothing about gstreamer but a BambuTunnel::GetMsg(1)
Are there any options to enable gstreamer logs?
I'll post another log pack with debug=5, there are some MediaPlayCtrl lines while I clicked PLAY button. Bambu_studio_debug.tar.gzhttps://github.com/user-attachments/files/16930067/Bambu_studio_debug.tar.gz

eintr, 2024-09-11:
This is another run.
`GST_DEBUG=4 Bambu_Studio_ubuntu-v01.09.05.51-20.04.AppImage --debug 5 >console.log 2>&1`

Seems much more better meaningful!

[console.log.gz](https://github.com/user-attachments/files/16963353/console.log.gz)

eintr, 2024-10-22:
I got some clue: I locked the screen after printing was started yesterday. And today morning, I woke up the screen and found it was playing the video of last night normally, maybe from the very moment I locked the screen.

So, I guess it is a bug about buffer underflow.

eintr, 2024-10-30:
Gone in version 1.9.7.52

closed 2024-10-30 by eintr