bambulab/BambuStudio #4788: Sending File to Printer Errors

JordanFa13, 2024-09-09:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

WIndows 10

### Printer

Bambu X1-Carbon

### How to reproduce

1. Send sliced file
2. send

### Actual results

when sending print files, just using the 'send' or 'send all' options give me errors for not being connected to the printer but when I use the 'print plate' option, the printer receives the file and prints it.

### Expected results

Printer should receive sent files and not give me an error that is circumvented by using the 'print plate' command

### Project file & Debug log uploads

[slim packout 4 bay divider x8.zip](https://github.com/user-attachments/files/16935639/slim.packout.4.bay.divider.x8.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-09-10:
Thanks for your feedback, this feature is under development and will be supported in the next release

JordanFa13, 2024-09-10:
This has always been a feature. The issue is sending the sliced plates to
the printer pop up with a LAN error but the print plate doesn’t.

walterwongbbl, 2024-09-10:
@JordanFa13 
This is a problem because send relies heavily on FTP transmission, which is highly dependent on network quality.

We are optimizing this feature and it will be effective in the next release version.

JordanFa13, 2024-09-10:
> This is a issue because send relies heavily on FTP transmission, which is

So why does the print plate option work instead?

walterwongbbl, 2024-10-15:
@JordanFa13  
 print plate uses our cloud service for printing.