bambulab/BambuStudio #4791: Bambu lab cloud no response !!!!

Stefbull, 2024-09-10:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

win11 64bit

### Additional system information

i9-rtx4070-32gbram

### Printer

X1c

### How to reproduce

[image]

### Actual results

?

### Expected results

?

### Project file & Debug log uploads

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fluppir, 2024-09-10:
Yes me too - looks like whatever cloud maintenance they had, has failed some.

ttt888zzz, 2024-09-11:
Yesterday, the official U.S. server migrate moved from California to Oregon, and a system anomaly occurred between 17:08 and 20:36, causing access problems.Live announcements were posted on [status.bambulab.com](http://status.bambulab.com/)。
Please try again. We are very sorry for the inconvenience。