bambulab/BambuStudio #4796: Problem with AMS Lite detection by Bambu Lab A1 Mini printer

ChickChickenKFC, 2024-09-10:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 mini Combo

### How to reproduce

When the printer is turned on but does not print, remove the cable from the AMS lite

### Actual results

I have a problem with my Bambu Lab A1 Mini printer, which stopped detecting AMS Lite after accidentally disconnecting it. I have already tried several solutions, but unfortunately the printer still does not recognize AMS. Here are the details of the problem:

Printer model: Bambu Lab A1 Mini
AMS Lite works properly - mechanically and in terms of connections, but the printer does not detect it.
I tried restarting both the printer and AMS Lite, checked cable connections, ports and updated the printer firmware, but to no avail.
I made sure the cable was fully functional and the ports were clean.
The printer does not show any AMS-related errors, but there is no option to manually detect this device.
ams pulls in the filament normally and 4 white LEDs light up
I would like to ask for help in solving the problem or for advice on further steps I can take.
[image]

[image]

### Expected results

The printer does not detect AMS

### Project file & Debug log uploads

[Uploading debug_Tue_Sep_10_15_23_28_2956.log.0.zip…]()

### Checklist of files to include

- [X] Log file
- [ ] Project file

ChickChickenKFC, 2024-09-10:
I forgot to add that I'm new to printers and I'm not very good at writing on forums, so if you have any questions, feel free to write.

gaolegao-lx, 2024-09-12:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

closed 2024-09-12 by ChickChickenKFC

TWhynot, 2024-12-11:
Having the same issue. What was the fix?