bambulab/BambuStudio #4797: Error 07FF C003 - but there is no filament inside the tubes

Bumblebiber, 2024-09-11:
### Bambu Studio Version

Irrelevant

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

BambuLab X1E

### How to reproduce

Happens when I try to load/unload filament

### Actual results

Error occurs. I've checked all tubes (also inside AMS) if there's any piece of filament stuck. But all tubes are free.

It might be worth mentioning that I did a maintenance last week and after that the error occured.

A firmware update didn't help either.

### Expected results

No errors

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gaolegao-lx, 2024-09-13:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Bumblebiber, 2024-09-13:
Done. Ticket No: US240913323001

larsony99, 2025-07-03:
> Done. Ticket No: US240913323001

Was there ever a resolution for this? I am getting the same error now.