bambulab/BambuStudio #4798: Manual tree support generating bug

GoldhandRenthal, 2024-09-11:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Printer

Bambu Lab X1 Carbon

### How to reproduce

T34 tank model "belly" armor's tree support added via the "fill tool".
1st layer OK
it skipped the 2nd, 3rd 4th 5th layers
starting the tree branch again from the 6th layer
[image]
[image]

### Actual results

the mentioned tree supports are messy from the 2nd layer

### Expected results

it should not skip 5 layers of support, and expect to lay the 6th layer fine LOL

### Project file & Debug log uploads

[Tank_95mm_Tiger_and_T34_BUG.zip](https://github.com/user-attachments/files/16962032/Tank_95mm_Tiger_and_T34_BUG.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

GoldhandRenthal, 2024-09-12:
FYI, all the models, even the t34 "belly" turned out great at the end.
I was a bit worried and sceptical around the 6-10th layers because of the issue mentioned above.

tangketan, 2024-09-13:
This is a known issue with organic support. Please change support style to "tree hybrid" instead.