bambulab/BambuStudio #4800: win版本Bambu Studio的图标下面脏脏的，真的不是我屏幕问题

LaowuENT, 2024-09-11:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

win10 22H2

### Additional system information

amd 5800H 
3070 laptop 
2k165hz屏幕

### Printer

p1s

### How to reproduce

使用2k屏幕开125%缩放的时候，桌面图标下方脏脏的，在2台设备复现，1080p无此情况

### Actual results

[image]
就这样

### Expected results

[image]

### Project file & Debug log uploads

这我就不写了吧

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LaowuENT, 2024-09-11:
“项目文件和调试日志上传
这我就不写错了”

打错了，是：“这我就不写了”

ttt888zzz, 2024-09-13:
感谢您的反馈，该问题已经记录，我们会跟进一下