bambulab/BambuStudio #4801: Bambu Studio 1.9.5.51 Keeps Crashing when slicing

camcooler, 2024-09-12:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Edition	Windows 11 Pro
Version	23H2
Installed on	‎2/‎2/‎2024
OS build	22631.4112

Processor	Intel(R) Core(TM) i9-14900KF   3.20 GHz
Installed RAM	96.0 GB (95.8 GB usable)
CORSAIR Dominator Titanium RGB DDR5 RAM 96GB (2x48GB) DDR5 6600MHz CL32 Intel XMP iCUE Compatible Computer Memory - Black (CMP96GX5M2B6600C32)

System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

MSI PRO Z790-A MAX WiFi ProSeries Motherboard (Supports 12th/13th/14th Gen Intel Processors, LGA 1700, DDR5, PCIe 5.0, M.2, SATA 6G, 2.5Gbps LAN, USB 3.2 Gen2, Wi-Fi 7, Bluetooth 5.4, ATX)

ASUS Dual GeForce® RTX 4070 OC Edition 12GB GDDR6X, IP5X, Auto-Extreme Technology, 144-Hour Validation Program, HDMI 2.1a, DP 1.4a

### Printer

X1 Carbon

### How to reproduce

When slicing model, the status bar gets to about 80%, Generating G-code then the program crashes and closes completely. 

### Actual results

Bambu Studio crashes and closes completely. 

### Expected results

It should produce the G-code for printing. This happens to many of my files when I try to slice on this PC.  Once out of every 10 try's,  it will slice the file correctly. When I slice the file on a lower quality PC, it works as expected. You need to get this program to work with higher end computers. 

### Project file & Debug log uploads

[crash_Wed_Sep_11_20_11_55_0_done.log](https://github.com/user-attachments/files/16972878/crash_Wed_Sep_11_20_11_55_0_done.log)
[Netgear Nighthawk 6 Pro Cooler v26.zip](https://github.com/user-attachments/files/16972898/Netgear.Nighthawk.6.Pro.Cooler.v26.zip)

[crash_Wed_Sep_11_20_11_38_0_done.zip](https://github.com/user-attachments/files/16972894/crash_Wed_Sep_11_20_11_38_0_done.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-09-13:
HI，[camcooler](https://github.com/camcooler)：
Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

Athomiser85, 2024-11-02:
Hello,
On Bambu Studio version 1.9.7.52, from official website, I still have the issue, even after trying the temporary fix you mentioned. 
Here are my specs : 

- Windows 11 23H2
- AMD Ryzen 9 7900X
- Nvidia RTX A2000 12GB
- 32 GB RAM

I tryed with several core values, 4,7,8,24 Core .

In order to find a workaround, I downloaded and installed the last version of the software V1.10.0.81 ( Beta).
Then tried slicing with only 8 core affinity and without modifying. The software still crashes, but just after the slicing finished.

It used to work with an older version, 6 month ago from now (don't remember wich one).

If it helps, here the content of crash_Sat_Nov_02_10_02_51_0_done file : 
`Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x1 
Param 1 :0xa845e000 
Context :000000E2E9CEBA00 
ContextFlag : 0x10005f, EFlags: 0x10202 
Fault address:  0xA5B3491B 0x0:0x0 

Show CallStack:
`

Jamrr, 2024-12-24:
This is happening to me too even after the fix shared above.

Flanchaves, 2025-04-22:
Im also having the error