bambulab/BambuStudio #4804: Object hits the X-axis of A1 Mini printer when "By object" slicing selected - SW does not warn properly.

ckupcu, 2024-09-12:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.6.1 

### Additional system information

Apple M1 Pro

### Printer

BambuLab A1 Mini

### How to reproduce

1. Load the attached file
2. Select Print Sequence "By Object"
3. The first tall object printed at the back is hitting X-Axis metal when second object starts printing...

### Actual results

When you move the second object by mouse manually, SW shows circles to indicate intersection zones. But this is not sufficient, SW should take care also horizontal clearance along the X-axis and warn user...

Said that, Auto Arrange button calculates positions correctly so that no collision happens.

### Expected results

The SW should show not only circular avoidance zones, but also show horizontal avoidance zone for X-axis collision when manually moving objects on the build plate.

See attached screen snapshow, this warning does not show up when you load .3mf file. Zone is not calculated properly...

[image]

### Project file & Debug log uploads

[Cover, base cover, reflector - V2.3mf.zip](https://github.com/user-attachments/files/16977130/Cover.base.cover.reflector.-.V2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ckupcu, 2024-09-14:
Hi Olekshar, 

Yes, on the picture it is the moment when it gives the error - 2nd object on the left is moved to show the error. But in the test file I attached, object is still beyond that "hitting" limit. And if you print it, it hits the X-Axis. I suspect, SW does not calculate the hitting limit properly.
Try loading the file and print, you will see.

oleksharh, 2024-09-14:
I have adjusted the printing order to go from front to back. The file is attached, and it should function properly. However, you're right that it doesn't account for everything, and it doesn't rearrange the order of printed objects to avoid collisions along the x-axis.
[Cover_Base_Reflector.zip](https://github.com/user-attachments/files/17002186/Cover_Base_Reflector.zip)

ckupcu, 2024-09-14:
Hi Oleksharh,

Do you plan to fix this in a SW update?

oleksharh, 2024-09-14:
Hi ckupcu,

I will have a look on the next week

lanewei120, 2024-09-16:
we can reproduce it on our side too
and we doubt the radius used by A1 mini has some deviation
we will check it soon

ckupcu, 2024-09-16:
Great!

ckupcu, 2024-10-08:
Any progress on this issue?

ckupcu, 2024-11-14:
Hallo, got installed today official v 1.10.0.89. I tested this problem, looks like the App is warning much better than before. I did try few other files with challnging objects, seems to be working now.

Besides, if you go to Preview with a bad placement, it is not slicing the tray. Which was not the case before.

I have managed some manual placements myself taking care about height and position assuming it would print, but it is more pesimisticly rejecting now - which is acceptable from side.

Thank you for fixing.

lanewei120, 2024-11-21:
already fixed in 1.10.0

closed 2024-11-21 by lanewei120