bambulab/BambuStudio #4805: Cannot set coordinates

stefek47, 2024-09-12:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Add object,
2. Add negative part
3. Set negative part X coordinates 
4. Set negative part Y coordinates

### Actual results

When I set the Y or Z coordinates, it moves in different direction and the set coordinates doesn't change.
When I use the move options, arrows are in wrong places.

### Expected results

Object should move to set coodinates.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16978492/log.zip)
[adapter.zip](https://github.com/user-attachments/files/16978494/adapter.zip)
[image]
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-09-13:
Hello, can you provide 3mf? Thank you~

Haidiye00, 2024-09-13:
[image]
We have added a local coordinate system in the internal version, please wait for the next version

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00