bambulab/BambuStudio #4809: Height range Modifier settings are not copied on Fill bed with copies

longtimeago, 2024-09-12:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04.4

### Printer

Bambulab X1C Carbon Combo

### How to reproduce

1. Select model
2. Add "Height range Modifier"
3. Adjust Layer Height to value different from default
4. Select model
5. Click "Fill bed with copies"

### Actual results

Height range Modifier is not copied

### Expected results

Height range Modifier is preserved and copied

### Project file & Debug log uploads

Project file: [proj.zip](https://github.com/user-attachments/files/16983173/proj.zip)
Log file: [debug_logs.zip](https://github.com/user-attachments/files/16983193/debug_logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

longtimeago, 2024-09-30:
@zhimin-zeng-bambulab well, simple "Clone" has not performance problem copying all modifiers.

Even if "Fill bed with copies" will be slow it kinda expected and OK.
What is not expected is to silently skip copying modifiers.

Thus, I see 2 options:
1. Make "Fill bed with copies" to copy modifiers with all attributes and accept that it may be slow (preferable behavior)
2. Make it super explicit that modifiers are not copied on "Fill bed with copies" (to avoid surprises and further issues)

wdyt?

dimitarsp, 2024-12-05:
we need real instancing, that's the issue. Clone is a simple duplicate, not an instance with shared settings