bambulab/BambuStudio #4811: Camera Feed Stops

MasterMatthew, 2024-09-13:
### Bambu Studio Version

1.9.5.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Labs X1C

### How to reproduce

Step 1: Open Studio
Step 2: Slice
Step 3 Send to pronter
step 4: Watch Live view

### Actual results

while watching the printer feed, after 15 mins or so the camera screen goes black and there is a message that says : 
 temporarily closed because there is no operating for a long time

### Expected results

be able to use the camera to monitor prints

### Project file & Debug log uploads

[debug_Fri_Aug_16_19_23_53_5804.log.zip](https://github.com/user-attachments/files/16988741/debug_Fri_Aug_16_19_23_53_5804.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-09-14:
The same issue https://github.com/bambulab/BambuStudio/issues/4690

tinman31k, 2024-09-15:
Any idea when the next update will be?  It would be great if it is soon so that the camera issue would be fixed.

MasterMatthew, 2024-09-15:
My work around. I installed older version of studio, wasn't having any trouble with it. So when it asked me to update, I selected 'Skip This Release'.

ttt888zzz, 2024-09-18:
HI [MasterMatthew](https://github.com/MasterMatthew)、[attochron](https://github.com/attochron)、[tinman31k](https://github.com/tinman31k)：The internal version has added a new configuration to control the automatic camera stop, and we will update this feature in the next version soon

MasterMatthew, 2024-09-18:
> HI [MasterMatthew](https://github.com/MasterMatthew)、[attochron](https://github.com/attochron)、[tinman31k](https://github.com/tinman31k)：The internal version has added a new configuration to control the automatic camera stop, and we will update this feature in the next version soon

Awesome! Looking forward to it. Much appreciated.
Thank you and tell the team thank you!

ttt888zzz, 2024-10-11:
HI[KedziePanda](https://github.com/KedziePanda)、[mupke475](https://github.com/mupke475): 
1.Please download the latest version and visit the website，https://bambulab.cn/zh-cn/download/studio
2.Update 【References】setting，according the photo
[image]
https://github.com/bambulab/BambuStudio/releases

closed 2024-10-11 by ttt888zzz