bambulab/BambuStudio #4812: "Color Scheme" panel is in an inconvenient spot

rgov, 2024-09-13:
### Bambu Studio Version

1.95.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia

### Printer

Bambu Labs P1S

### How to reproduce

X

### Actual results

Y

### Expected results

Z

### Project file & Debug log uploads

W

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Muegic, 2024-10-08:
Just started with Bambu Studio and this color scheme windows drives me crazy. Every time you slice it pops up again and blocks your entire view. Please give us the option to collapse it permanentely.

md81544, 2024-10-23:
I also would love perhaps an option in settings to keep this panel minimized by default.

BrianJPG, 2025-01-09:
This is still an issue. It annoys me every time I click "Slice plate."