bambulab/BambuStudio #4818: Concentric sparse infill generator is insufficient

onthuhlist, 2024-09-13:
### Bambu Studio Version

Absolute latest

### Where is the application from?

Bambu Lab Official website

### OS version

Absolute latest

### Additional system information

Mac Mini M1

### Printer

Bambu Lab A1 mini, A1, everything (Bambu Studio)

### How to reproduce

Bambu Studio has an issue with some models and concentric infill for clear filaments like PETG. See the attached model. Bambu Studio does not generate a good enough support to actually print this in real life. The sparse infill is loose like a limp rag, and just becomes a tangled ball of filament.

### Actual results

See above

### Expected results

Concentric sparse infill should be anchored well enough to not wiggle and flop around inside during printing. See attached .3mf model.

### Project file & Debug log uploads

[Uploading square mobius strip - concentric.3mf.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-09-14:
HI，[onthuhlist](https://github.com/onthuhlist)
Could you please provide the source file,we will check it.

ttt888zzz, 2024-09-18:
HI [onthuhlist](https://github.com/onthuhlist): After several attempts, the file you uploaded could not be downloaded.
[image]