bambulab/BambuStudio #4819: Cancel object from print when one object fails but others are fine

Dpariet, 2024-09-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I am printing an assembly of different parts and have 4 different objects printing on the same plate.  1 object has failed and the other 3 are printing perfectly.  I am now wasting all the filament and time on the failed object.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
On bambu studio or the handy mobile app, when you start a print have a screen that shows the print bed and objects being printed and have the ability to cancel an object out of the print if it fails sometime during the print.  (Sonic Pad has this feature)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
This is the only efficient solution

**Additional context**
Add any other context or screenshots about the feature request here.

DrymarchonShaun, 2024-09-15:
From the way you worded it, I can't tell if you are aware or not that this *can* be done from the mobile app - https://wiki.bambulab.com/en/general/skipping-objects#operate-on-the-app

Dpariet, 2024-09-16:
I did not know that, bambu studio should get this feature also!

thebadluck, 2024-10-13:
I agree this feature would be very helpful in Bambu Studio.  I have a P1S so the screen is useless for most anything.   I do not always have my phone handy when I am working at my computer.  Even when I do, it takes time to open the app.  I have seen this feature requested several times.

jlo1975, 2024-11-20:
we can do everything in bambu studio except this thing.....
So, yes not have the option in bambu studio to mark failed object to be skipped is a huge design mistake.

darterdave, 2024-11-26:
I just tried to save dozens of hours in a failing print. Used BH app for my P1S and skipped the objects that I wanted to cancel. It was confirmed. And printer continued to try and print the objects that I had just canceled. What a waste of time and money. Thousands of dollars of equipment and I can only abort the entire print job. Please make a solution to this issue.

prdoyle, 2025-01-18:
Dupe of #3281?