bambulab/BambuStudio #4823: Alternate extra wall

weenan, 2024-09-15:
Weakest part of a heated insert is the bond between the infill and the walls of the hole for the insert. Insert pulls out together with the walls separated from the infill.

Orca slicer has a feature called "Alternate extra wall" which prints one extra wall every other layer so the infill is sandwiched in with the wall. This greatly improves the bond between the infill and the wall.

Would be great to have this feature in Bambu Studio.

[image]

jensim, 2025-05-05:
This feature is amazing! It's enough to have me switch to Orca for prints that require additional strength
One might call this **infill brick layering**, as it gives the print a horizontal brick layering pattern between the infill and wall
It's just such a strong thing, especially when I print with fat nozzles, 1 wall with alternate extra wall is enough

Hammerfest, 2025-05-13:
Yep, its one of the primary reasons I use OS instead of BBS along with support interface fan control

Onecton, 2025-05-16:
Cura Slicer has this also dor years now. I really miss that feature in BambuStiudio. I second this!

DiCaprio37, 2026-04-24:
I'd love this feature as well on linux

teafusion, 2026-04-25:
> > Bambu Studio has never had 'Alternate (Extra?) Wall' feature to my reckon, at least not by that name, but from the 'Strength' tab you can still find 'Embedding the wall into the infill' which is Bambu Studio's version of the feature and found right under the 'Wall Loops'. You don't even need to switch on the 'Advanced' mode to be able to use it. Hope this helps!
> 
> this is most certainly not doing the same thing as alternate extra wall. Watch https://youtu.be/c7CI6yBTKMc?si=zdEVMeCAi9bFEaMo to see it is applied to all the wall/infill spots. "Embedding the wall" setting is not doing the same thing:

Thank you for your input. Well acknowledged that it's not the same, yet when the 'Embed wall into infill' was introduced in Bambu Studio late last year, it was supposed to be the Bambu Studio substitute for the 'Alternate Extra Wall', without any explanations why that would be so, rather than the actual feature requested here. But lucky we found a guy to port it to Bambu Studio, hope it catches on among the core team eventually.

sbrocket, 2026-09-09:
FYI for others who might otherwise comment here about it: @teafusion filed #11109 about adjusting the pop-up warning