bambulab/BambuStudio #4825: Using the Fuzzy Shell option will create patterned textures on the surface.

ANicksn, 2024-09-16:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lan X1C

### How to reproduce

1) Go to the "other" tab
2) enable the "fuzzy skin" mode
3) leave the default settings
4) start printing

### Actual results

After updating to the latest version, vertical stripes began to appear on the surface, i.e. the fuzzy shell became less random (apparently the fix in version 1.9.4 no longer works and repeating patterns appear again).
[image]

### Expected results

Expected to get a random fuzzy surface, as it was before update 1.9.5.51
[image]

### Project file & Debug log uploads

[Абажур_новый.zip](https://github.com/user-attachments/files/17011043/_.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-09-18:
Thank you for your feedback. The internal version has been fixed and we will fix it in the next version.

ttt888zzz, 2024-09-20:
HI [ANicksn](https://github.com/ANicksn)：Please download the latest version, as the issue has been fixed.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz