bambulab/BambuStudio #4827: Smooth coefficient not working all the time. Abrupt speed changes still exist

goodman-capu, 2024-09-16:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.6

### Additional system information

MacBook Pro, Apple M1 Pro chip

### Printer

Bambulab A1

### How to reproduce

* Use the 3mf file in https://forum.bambulab.com/t/holes-with-petg-hf/97727
* Set smooth coefficient to a very low value, like 0.1.
* Slice it.

### Actual results

Some abrupt speed change on certain layers (not all)
https://forum.bambulab.com/t/holes-with-petg-hf/97727/10

### Expected results

Smooth speed change on all layers. (Since smooth coefficient is set)

### Project file & Debug log uploads

[joystick_mount_lower_adapter_v7.0_test_PSta_MBaPETGHF_N4.3mf.zip](https://github.com/user-attachments/files/17020090/joystick_mount_lower_adapter_v7.0_test_PSta_MBaPETGHF_N4.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

goodman-capu, 2024-09-17:
This is a simpler 3mf file that [demonstrates the same issue](https://forum.bambulab.com/t/holes-with-petg-hf/97727/12?u=gm_frank) (credit to model author GuiltySpark343):
[joystick_mount_lower_adapter_v7.0_test2.3mf.zip](https://github.com/user-attachments/files/17033454/joystick_mount_lower_adapter_v7.0_test2.3mf.zip)

QingZhangBambu, 2024-09-18:
yes. need to improve. thanks to report that.

GuiltySpark87, 2024-10-02:
I second this. Is there a rough timeline as to when we can expext a fix?

fahaaaxiki, 2024-11-05:
[image]
@GuiltySpark87 @goodman-capu 1.10 Beta2 (https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81) has updated preset and this problem (PETG HF) could be handled like pictures above.

MartinChan3, 2024-11-06:
I print another one with orange Bambu PETG-HF in Studio 1.10 Beta2  ( https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81 ) 
[image]

goodman-capu, 2024-11-06:
> I print another one with orange Bambu PETG-HF in my A1(0.4 nozzle), Studio 1.10 Beta2 ( https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81 ) ![img_v3_02gc_211cef91-a5e7-49f1-9cd4-bba1f73d00fg](https://private-user-images.githubusercontent.com/19528955/383394096-b65aaa9b-da7e-4303-b400-df55ee6d35cc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NjcyNzgsIm5iZiI6MTczMDg2Njk3OCwicGF0aCI6Ii8xOTUyODk1NS8zODMzOTQwOTYtYjY1YWFhOWItZGE3ZS00MzAzLWI0MDAtZGY1NWVlNmQzNWNjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA2VDA0MjI1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVkNjQyNTQ1YmVkZTNhZDQzOTE4N2U0ODkyOGIzZmEyZjQ3YjYzOTE3MzZjNGNmNmI2NzY3M2MxMjk2NWM2NDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.96J6uEu1L5Qvj5WczgwMy-WKo_6moIKWL52v3-QLybY)

Seems like the issue was not fixed :(

QingZhangBambu, 2025-03-03:
I think it is what you want
[image]

goodman-capu, 2025-03-03:
> I think it is what you want [@goodman-capu](https://github.com/goodman-capu) [image]

Yes right side is the intended behavior. Will it be available in the next release? I tried 1.10.2.76 and still get the left result.

QingZhangBambu, 2025-03-03:
Internal testing is needed, and the next version may not be in time. I will discuss this with the product manager.

QingZhangBambu, 2025-03-03:
if any news, i will tell you