bambulab/BambuStudio #4828: Cannot transfer files to printer whiel in LAN Mode

CyberGWJ, 2024-09-16:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

WIndows 11

### Printer

Bambu Lab X1C with AMX

### How to reproduce

1. Get MakerChip_Collector_Capsule_v1.3mf from K2_Kevin on MakerWorld.  
2. Slice 
3. Send \ Print Plate - gets error

If I disable Window Firewall the transfers work.

I have only had the Bambu X1C for 2 days and it is my first Bambu printer.  It is not a very good experience and after spending ~ $2000.00 for a X1C and 2 AMS I am NOT happy.

### Actual results

The progress bar on the transfer goes so far then it hangs (see videos).   

### Expected results

Transfer of file to printer to be successful.

### Project file & Debug log uploads

[Bambu Studio.zip](https://github.com/user-attachments/files/17021046/Bambu.Studio.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-09-18:
Please check the setting of the router or your PC to make sure the port is not blocked,https://wiki.bambulab.com/en/general/printer-network-ports
[image]

CyberGWJ, 2024-09-18:
@ttt888zzz Thanks.  I was looking for what ports it was using.  Although thoses ports are open. I can use WinSCP to connect using FTP on port 990.  Sometimes it will copy the file and at 70% it hangs.  Other times it hangs at 30%.  I was lucky ONCE and it copied the file and started the print. I install Microsoft NETMON and did packet captures on the transfer.  Everything looks good. Other than the X1C doing a lot of chatting with various time servers (port 123).  I am wondering if the problem is on the X1C itself.  Some how it is failing to communicate back to Bambu Studio.  The X1C is on the same network subnet as the computer.

TennisRunner, 2024-09-18:
> @ttt888zzz Thanks. I was looking for what ports it was using. Although thoses ports are open. I can use WinSCP to connect using FTP on port 990. Sometimes it will copy the file and at 70% it hangs. Other times it hangs at 30%. I was lucky ONCE and it copied the file and started the print. I install Microsoft NETMON and did packet captures on the transfer. Everything looks good. Other than the X1C doing a lot of chatting with various time servers (port 123). I am wondering if the problem is on the X1C itself. Some how it is failing to communicate back to Bambu Studio. The X1C is on the same network subnet as the computer.
> 
> Note: With WinSCP I can copy files back and forth with no issues.

Try updating the printer's firmware to the latest version using the sd card. I think I had a similar issue a while ago. 
Also make sure your printer has a clear line of sight to the router in case its a intermittent signal issue.

CyberGWJ, 2024-09-18:
@TennisRunner thanks for the idea.  I checked the firmware and it is up to date 01.08.02 for the X1C.  The printer is about 10 feet from the access point.  There is wall between them.  Normal house construction 2x4 and drywall.  No major electrial equipment between them. 

This is frustrating.  With my other 3D printers I have a lot of ways to troubleshoot the issue.  I am not use to a closed system!

TennisRunner, 2024-09-18:
Try running it with bambu studio completely whitelisted in the firewall. If it goes through without an issues. Then you'll know for sure its an issue with bambu studio and not your internet or printer.

CyberGWJ, 2024-09-18:
I have whitelisted bambu studio already and my account on this pc has administrator rights.  This pc is about 15 feet from the printer and access point.  I will try another pc.  It will take time to configure it.  I am puzzled that WinSCP works without any issues and bambu studio doesn't.

TennisRunner, 2024-09-18:
> I have whitelisted bambu studio already and my account on this pc has administrator rights. This pc is about 15 feet from the printer and access point. I will try another pc. It will take time to configure it. I am puzzled that WinSCP works without any issues and bambu studio doesn't.

I thought you only whitelisted certain ports instead of all?

CyberGWJ, 2024-09-18:
@TennisRunner I started with ports the went to app.  I at a loss what to try next short of disabling the firewall everytime a file needs to sent to the printer.

TennisRunner, 2024-09-18:
> @TennisRunner I started with ports the went to app. I at a loss what to try next short of disabling the firewall everytime a file needs to sent to the printer.

Have you confirmed that the issue only appears when you have the firewall enabled?

CyberGWJ, 2024-09-18:
@TennisRunner Yes. My process is to have the firewall enabled as normal.  Then send a file to the printer.  When it fails, turn off the firewall and resend.  It always works.  Then trun the firewall back on when the transfer is complete.

TennisRunner, 2024-09-18:
If you send it a second time without changing the firewall.
Does it fail every time? If no, how often?

As a side test, try going to the device tab, selecting your printer.
Then moving the printer head or toggling the lamp.
Maybe doing that will make it sync/handshake better before you send the file.

Also I noticed its saying its doing a timeout on the FTP connection.
Time how long it takes for it to fail.
If it stops at the same time, every time. Then it may be hitting the total timeout allowed rather than an intermittent connection issue. 
How long does it usually take to do that timeout error?

I also noticed that its doing that error message in the preview window instead of switching to the device tab and then showing the upload progress.
Are you slicing the model in the printer and then printing that or loading a .gcode file and printing that?
Does it have the issue if you slice the stl/step file yourself?
Does it have this timeout issue on other models?
Does it have this issue with small models?
I just tested a 5 inch model and it only takes 0.3mb and transfers in 10 seconds.

In your original post you say "see videos", but I only see 2 screenshots. Are there videos of this issue?
In the log file attached, what percent did it time out on?

I'm throwing wild guesses.

CyberGWJ, 2024-09-18:
@TennisRunner It fails everytime I resend the file.

Device Tab.  I will try that after my current print finishes.  

It takes at 30 seconds to timeout and it is consistant with the timeouts on each send.

I alway slicing the stl in Bambu Studio then doing the transfer.  I open BS and copy a stl, make changes then send.  Or I open a previous project slice then send.
All models seem to timeout.  I have only had the printer for a few days.  So, I am just getting things working.
Model size doesn't seem to matter so far.  I have printed both small and large.
When it starts the transfer the copy goes fast.  Then it timesout.

I tried to send the videos but they are large files and github would not allow it.  
70%

Thanks for your help.  I feel that I am to focused.  I am retired and worked in the IT world for a long time.  Many roles (sys admin, net mgr, etc) I have not seen anything like this before.

TennisRunner, 2024-09-18:
Maybe they need to use passive mode on the FTP transfer settings.
They probably won't be able to replicate this issue unless you give a detailed breakdown of what you're doing to block bambu studio and even then since its third party interference, they may not prioritize trying to fix it. 
If you want to speed unblocking and reblocking it. You should probably do a hotkey, macro, script that saves you time.
Probably unrelated, but you could try a different SD card in case its a bad sector issue.

CyberGWJ, 2024-09-18:
@TennisRunner In LAN Mode passive FTP should be fine.  Since I don't know what is causing the block it will be hard to figure out.  I may have to look at doing port mirroring on the switch to see all the packet.  But that is a lot of work to setup and analyse.  The big problem I cannot see what the X1C is doing.  

I know other people are having the same issue so I'm not alone.  

I have already looked at the hotkey option and will probably do that for the time being.  I have a few more SD cards and will try them.  

This will all have to wait until tomorrow.  I am done fighting this for the day.  Again, thanks for assistance.

CyberGWJ, 2024-09-19:
@TennisRunner Just an update.  From Bambu Studio I can see videos and home the printer.  There was also an update to Bambu Studio today.  It didn't solve the problem.  

I don't think the problem is within Bambu Studio rather the network add-on that gets installed.  This add-on is for network connectivity to printer.  

I have given up on trying to figure out what the problem is.  There is no way to see what the X1C is doing when transferring a file.  Or what the add-on it doing.  So, I created a powershell script and assigned it a hotkey.  

If the printer did not do multi-color well. I would have returned it.  It unacceptable for Bambu Labs not to be more open about it's print environment.  I will not recommend it to anyone that needs to LAN Mode.

Regards

walterwongbbl, 2024-10-17:
@CyberGWJ @TennisRunner 
We have found that the FTP experience is not good, and we will give up using FTP in the future and develop new file transfer functions ourselves.

CyberGWJ, 2024-10-17:
@walterwongbbl Thanks for the update.  I look forward to when it is released.

jarrard, 2024-12-14:
Problem persists. Can't even get printer added in lan mode even when detected.

CyberGWJ, 2025-03-26:
Is there any update on this issue?  It is really annoying to have to disable the firewall each time I need to print something!

skintigh, 2025-03-27:
@CyberGWJ can you share your scripts, please?

2 years later and Bambulab hasn't lifted a finger to fix LAN-only mode, they are too busy locking the printer down even more and making it work even less. 

I would have upgraded the H2D had my experience with Bambu's basic functionality and reliability not been so terrible.

CyberGWJ, 2025-03-28:
Here is the PowerShell script.  I got the basic from somewhere and updated it to suit my needs.

#
# Toggle Domain, Public, Private firewalls 
# This for Bambu Studio network transfer issues
#
$FWProfiles = (Get-NetFirewallProfile);
Write-Host "Windows Firewall Profile Statuses" -Foregroundcolor Yellow;
$FWProfiles | %{
    If($_.Enabled -eq 1){
        Write-Host "The Windows Firewall $($_.Name) profile is enabled - Turning Off!"  -Foregroundcolor Green
	Set-NetFirewallProfile -Profile $($_.Name) -Enabled False
        }Else{
        Write-Host "The Windows Firewall $($_.Name) profile is disabled - Turning On!" -Foregroundcolor Red
	Set-NetFirewallProfile -Profile $($_.Name) -Enabled True
        } 
    };

Pause

CyberGWJ, 2025-03-28:
Here is the actual file.  NOTE: change the file type from txt to ps1.

[firewall_on_off.txt](https://github.com/user-attachments/files/19511113/firewall_on_off.txt)

DragonFlyer007, 2025-05-04:
Same issue here, so fed up with this...
Even if specified ports are open, no change. Open all ports, no change.

Dissable firewall, works!
BUT i need the firewall for other stuff...!

Can also send thru WinSCP, no problem, Sooo... Must be on Bambu side the issue lies.
Sometimes it will send the file even if it times out on FTP. 

But yeah, Bambu need to get a grip on this ASAP, many people frustaded with this now.
Just ordered the H2D as well, but almost regretting because of stupid stuff like this interuprint every single print job..