bambulab/BambuStudio #4829: AppImage Not Launching

mattkwarren, 2024-09-17:
### Bambu Studio Version

1.9.4

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 40 Workstation

### Printer

N/A

### How to reproduce

1. Install AppImage
2. Launch AppImage

### Actual results

Application does not launch

### Expected results

Launch of application

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mattkwarren, 2024-09-17:
To clarify, I have also tried version 1.9.5 with the same results. Both AppImage files were labeled Fedora, which is unusual for AppImages since they are meant to run on any distro that supports AppImage. Could some non-standard packaging issue be the cause?

MackBambu, 2024-09-20:
Launching appimage in the command line and observe whether there is any error message?

danexpo, 2024-11-07:
Surprisingly the Fedora appimage doesn't work on Fedora 41, but the Ubuntu appimage does!

mattkwarren, 2024-11-20:
> Launching appimage in the command line and observe whether there is any error message?

There is a newer issue that may be relevant to this issue: https://github.com/bambulab/BambuStudio/issues/4877

thenhesaid, 2025-01-02:
> Surprisingly the Fedora appimage doesn't work on Fedora 41, but the Ubuntu appimage does!

YES I found this today as well. After I lost 1hr... I successfully installied the Ubuntu imageapp on my Fedora41_X64 machine :-/