bambulab/BambuStudio #4831: Login Failed (83968036)

TheSpiderDungeon, 2024-09-17:
### Bambu Studio Version

Doesn't matter

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

A1 Mini + AMS

### How to reproduce

1. Tap "account" in printer settings

### Actual results

QR code is in "Refreshing" endlessly and "Login failed (83968036) appears after some time.

### Expected results

I should get a QR code to login to my account so that I can pair my printer.

### Project file & Debug log uploads

Tell me how to get the logs from a printer I can't connect to, and I'll happily provide them.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TheSpiderDungeon, 2024-09-17:
I've tried separating the 2.4Ghz network into its own network (named pl_upward just in case that matters); I've tried rebooting the printer, the router, and my computer; I've tried formatting the SD card; I've tried toggling LAN only mode; and I've tried forgetting the network and reconnecting.

I can't find any more potential solutions online aside from setting up a "mobile hot spot" which I have never _not_ had trouble doing for any reason, so it has not been attempted.

gaolegao-lx, 2024-09-18:
**Please upload log through Handy APP, and give us your ticket number**

_**Upload log files through Bambu Handy**_
Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
To upload files, go to **Me → Support Tickets →** Choose the ticket for log upload and proceed with the operation.
[image]

TheSpiderDungeon, 2024-09-18:
I'd love to provide logs from the printer, but I can't even connect to it in the first place.

gaolegao-lx, 2024-09-18:
Connect the machine through nearby devices on Handy
[image]

TheSpiderDungeon, 2024-09-18:
The printer does not show up in normal or LAN only mode. Both devices are connected to the same network.

gaolegao-lx, 2024-09-18:
Can you help confirm if your printer's WiFi is connected properly, Such as clicking on the WLAN option on the printer screen can display the IP address normally. 
If WIFI is not a problem, please give me the logger folder in the SD card

TheSpiderDungeon, 2024-09-18:
IP is displayed as expected, but the SD card is throwing errors.

"D:\logger is not accessible. The file or directory is corrupted and unreadable."

Formatting it via the printer doesn't seem to have worked...

gaolegao-lx, 2024-09-18:
Please follow the above steps to provide feedback on your issue through Handy's printer ticket service. 
We will have a dedicated after-sales team to handle your problem. 

In the latest version of Handy, it is not necessary to connect a printer to use Handy's printer ticket service, and the machine SN can be manually output. 
[image]

gaolegao-lx, 2024-09-18:
You can first submit Handy's printer ticket service. It's okay if you can't upload the log temporarily, let the after-sales team handle your situation,
You can track the issue on the ticket service in the future,

gaolegao-lx, 2024-09-18:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

TheSpiderDungeon, 2024-09-18:
I was in the process of doing that as per your recommendation, but I can't send the ticket because it seems the inaccessible log files are required there, too.

TheSpiderDungeon, 2024-09-18:
Scratch that - there's no log requirement for listing it as an app malfunction. Not quite the topic, but it's close enough.

gaolegao-lx, 2024-09-18:
I asked for guidance from the after-sales team, you can give it a try
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

TheSpiderDungeon, 2024-09-18:
I formatted the microSD card on my computer to properly erase all data, and formatted it on the printer before trying to log in again just to make sure I've got the best chance of getting that log file. It was functional in the computer, but upon formatting via the printer, the entire card became corrupted. No luck getting that log file so far.

gaolegao-lx, 2024-09-18:
Whether a log is needed depends on the type of problem you have chosen, some require it, some do not.
Try the guidance above, which says how to submit a work order without a log.

DonSmilo, 2024-10-09:
I'm having the same issue. Printer shows this code in the account section of settings. P1S with AMS. It was working fine for several weeks until I turned it off and put in the AMS today. Using with iPhone 15 Pro on iOS 18.0.something (whatever's latest as of 2024-10-09)

Things I've tried:
- WiFi network is now split into 5 and 2.4GHz networks. I've connected to the 2.4GHz, naturally. 
- I've turned the printer on and off again. 
- I've logged out of the app and back in. 
- I've uninstalled the app and reinstalled it. 
- I've cleared the printer from my known bluetooth connections page. 
- I've restarted my phone. 
- I've turned wifi and bluetooth on and off again on my phone. 
- I've factory reset the printer. 

I cannot connect the printer to my account no matter what I do. It refuses to connect to WiFi and won't display a pairing code in the account screen to work with Studio.

DonSmilo, 2024-10-09:
@TheSpiderDungeon were you ever able to resolve the issue? If so, may I please ask what steps you took to do so?

TheSpiderDungeon, 2024-10-09:
I was; I filtered out the printer's network interface by mistake as I didn't recognize "espressif" at the time. Make sure your router isn't restricting access to that interface.

Once I removed the restriction everything started working again.

DonSmilo, 2024-10-10:
Thank you! Sadly my issue was not the same, since I matched the IP addresses early on and opened things up. 

Do you mind if I toss my eventual solution here for posterity? I also put it on Reddit. 
Please leave this up. There were no threads when I was trying to fix things. 

--------Reddit Post Below:---------

You must split your network into 2.4GHz alone. Do this using address on your router's back to access settings website. 
I was using ATT's fiber nonsense. bgw210 or something. 
Anyway, I made a guest network and then allowed WPA-1 (it used to be only WPA-2) and it works now...
Only took 3 hours...

Solution was the same as in a thread in Bambu Lab forums called "unable to connect to wifi" 77599. 
However, my problems were different. The printer would connect to wifi, but it would then say that binding failed on the printer screen and the iPhone would get stuck on the connecting circle indicator on step 2/3 or 2/4 when setting up printer. When I checked printers on network tab after that, my printer appeared both as WiFi connectable and Bluetooth. No matter which I used though, I would get either "Maybe not on the same LAN Connection timed out" or some other error. 
Documented on GitHub too. "I'm having the same issue. Printer shows this code in the account section of settings. P1S with AMS. It was working fine for several weeks until I turned it off and put in the AMS today. Using with iPhone 15 Pro on iOS 18.0.something (whatever's latest as of 2024-10-09)
Things I've tried:
WiFi network is now split into 5 and 2.4GHz networks. I've connected to the 2.4GHz, naturally.
I've turned the printer on and off again.
I've logged out of the app and back in.
I've uninstalled the app and reinstalled it.
I've cleared the printer from my known bluetooth connections page.
I've restarted my phone.
I've turned wifi and bluetooth on and off again on my phone.
I've factory reset the printer.
I cannot connect the printer to my account no matter what I do. It refuses to connect to WiFi and won't display a pairing code in the account screen to work with Studio."

TheSpiderDungeon, 2024-10-10:
That's just good internet etiquette, haha

I hope you find the problem, and that it helps others down the road. Good luck!

Pixel48, 2024-11-19:
Had same problem - in my case it was DNS

kingsie, 2024-12-22:
WTF! this 83968036 error problem is al over the internet. With no solution! I lost my original starling to a power supply failure. They sent me a new one that works everywhere except my p1s. No solutions. I've tried most things I have access to to. Nope. So what the fuck am I to do? Buy a new machine.... Prusa?

CptDaimox, 2025-01-02:
Anyone got a solution yet? Mine is definitely connected and I can connect it with Bambu Studio on Windows in LAN only mode but not with Bambu Handy

ysiegel29, 2025-02-02:
Same error could not connect so did a factory reset and now I have this error. Can't connect anymore.

Jack4223, 2025-03-06:
Try phone and Bambu 3D printer all are in the same 2.4G WIFI LAN, then Bambu Handy can access Bambu QR code and recognized printer, this method working for me.

ysiegel29, 2025-03-06:
It finally worked once again until it is not anymore. This printer is great but these connection issue (specifically with bambu studio) are so painful and frustrating, a huge waste of time.