bambulab/BambuStudio #4834: impossible to see any writing in bambustudio

terxo, 2024-09-18:
### Bambu Studio Version

1.09.05.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

any

### How to reproduce

i just start the application

### Actual results

impossible to see any writing in bambustudio

[image]
[image]
[image]

### Expected results

i'm expeting to see the page

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

TennisRunner, 2024-09-18:
Does it have that issue in dark mode?
[image]

Maybe the text is white, its missing the translations, or its unable to display it due to the accent characters

bambu123, 2024-10-11:
Please provide log files.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log