bambulab/BambuStudio #4838: linux版本无法安装，显示没有证书，客服让我怕从这里提问。

skyminrd, 2024-09-19:
### Bambu Studio Version

Bambu_Studio_ubuntu-v01.09.07.50-24.04

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Printer

A1

### How to reproduce

安装时会出现

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ZefengWang, 2024-11-03:
为什么要强制用户将语言摄制成英文？

ZefengWang, 2024-11-03:
我当前的语言环境就是中文，并不希望改成英文，而这个设置要求用户把语言强制改成英文，很不合理阿

MackBambu, 2024-11-03:
提示信息显示你选择的时en_GB，也就是英国英语，可能系统没有这个语言，尝试用命令行切换成en_US.UTF8美国英语，看看打开是否正常，排除语言问题。然后然在切回zh_CN.UTF8简体中文，看看是否恢复

ZefengWang, 2024-11-30:
你这个解释很离谱，你的这个软件应该自动读取系统的语言，根据语言自动适配用户主题，而不是系统中缺少语言要用户强制修改自己系统语言，这跟强迫用户吃屎，没区别。
我按照错误提示，语言改了英文，注销以后再重启就换成英文了，比吃了shi还恶心人，建议修改，这是个严重bug

MackBambu, 2025-03-28:
Duplicate of #6193

closed 2025-03-28 by MackBambu

ZefengWang, 2025-04-03:
> 提示信息显示你选择的时en_GB，也就是英国英语，可能系统没有这个语言，尝试用命令行切换成en_US.UTF8美国英语，看看打开是否正常，排除语言问题。然后然在切回zh_CN.UTF8简体中文，看看是否恢复

这个有没有可能就是你这里检测语言环境有问题？本来是中文，但是你检测到是英英，这个是问题的关键地方