bambulab/BambuStudio #4840: TPU Not Recognized in new Bambu Studio Update

Zaphodjoe, 2024-09-19:
### Bambu Studio Version

1.9.7.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 7 7800X3D 8-Core Processor              4.20 GHz
32.0 GB (31.2 GB usable)
64-bit operating system, x64-based processor

### Printer

Bambu Labs A1

### How to reproduce

I updated and refreshed my Filaments and it always brings them over from the printer. I have 4 spools on the AMS lite and TPU HF Set as external. When I do this now I get the error:
"There are some unknown or uncompatible filaments mapped to generic preset. Please update Bambu
Studio or restart Bambu Studio to check if there is an update to system presets."
and it sets my 5th filament as ABS and I can't even set it as TPU anymore. I checked my added filaments and it has TPU 95A and 95A HF Selected but it wont let me change it in Bambu Studio so I can't even print with TPU right now.

### Actual results

I can't use TPU anymore

### Expected results

I should be able to select TPU

### Project file & Debug log uploads

It doesn't matter the project. 
[Bambu lab TPU-HF profile (Benchy).zip](https://github.com/user-attachments/files/17063343/Bambu.lab.TPU-HF.profile.Benchy.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Zaphodjoe, 2024-09-19:
I ended up loading a TPU print Job and now it works and shows TPU. Someone in the Bambu Discord said the default nozzle could of been set to .2 or something TPU doesn't work with and thats why it wasn't working before. Now its loading though. I can't figure out how to delete this so please delete the bug report. Sorry for the confusion.

XunZhangBambu, 2024-09-26:
@Zaphodjoe ok，we will close the issue.

closed 2024-09-26 by XunZhangBambu