bambulab/BambuStudio #4841: X axis offset on multiple model when changing color with AMS lite

jmilhau, 2024-09-19:
### Bambu Studio Version

1.9.7.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.0

### Additional system information

Since today (since latest Studio update? v1.9.7.50) any print I make with several colors in the same layer gets a x-axis gap a several millimeters. Maintenance did not help.

[image]
[image]

Black was printed first, then purple then cyan then orange.

### Printer

Bambu Lab A1 Combo

### How to reproduce

Print a model with 2 colors on the same layer

### Actual results

Offset of the first color on the print
[image]

### Expected results

No offset
[image]

### Project file & Debug log uploads

[test v1.3mf.zip](https://github.com/user-attachments/files/17063455/test.v1.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jmilhau, 2024-09-19:
Re installing 1.9.5.51 solves the issue

BastR, 2024-09-19:
Same issue on A1, had to stop print. 
[image]

MCKanpolat, 2024-09-19:
I've just updated the latest version and same issue I have. Mac OS 15

BastR, 2024-09-19:
I did a few tests with Windows and printing went off without a problem.

ziehmon, 2024-09-19:
Are you using timelapse by any chance?

BastR, 2024-09-19:
In my case no, no timelapse on the A1

jmilhau, 2024-09-19:
No timelapse either

ziehmon, 2024-09-19:
Allright, can you do another check just to make sure: remove time timelapse_gcode in extruder settings? It's a wild guess but worth a try. The behavior was changed in the new release.

NicolasABSuarezB, 2024-09-20:
I have the same problem with mi A1 (Bambu Studio v1.9.7) on Windows
Solved with the reinstallation of Bambu Studio v1.9.5.51  

Don't give importance to the texture of the first layer, I'm trying some things
[image]
[image]

SaltWei, 2024-09-20:
[image]
Caused by this. Disable timelapse can workaround.

SaltWei, 2024-09-20:
Will revert this patch and hot fix this issue

BastR, 2024-09-20:
https://github.com/user-attachments/assets/eb58b82e-f6a8-4ead-9749-b1a6459cd841

@SaltWei @ziehmon As you can see from the video, even with the timelapse_code removed, there is still a layer offset.

You can hear the bed collide with the movement of the printer head, so I suppose that's when the step is lost and there's a shift.

XunZhangBambu, 2024-09-20:
@BastR To solve your issue currently, you need to turn off timelpase when printing rather than remove the timelapse gcode

jmilhau, 2024-09-20:
Timelapse was NEVER activated for me, so I do not think this solves the issue.

ziehmon, 2024-09-20:
@XunZhangBambu when the timelapse gcode is deleted, get_last_pos_from_gcode() would not change the position. The user also had timelapse disabled.

Where do you think the bug is exactly?

It must be happening when AMS is used, because users without one don't face this issue it appears.

BastR, 2024-09-20:
@XunZhangBambu As @jmilhau  says, the timelapse had never been activated when I made my prints, so I activated and deactivated the timelapse just in case, but I still have the shift.

ziehmon, 2024-09-20:
> @XunZhangBambu As @jmilhau  says, the timelapse had never been activated when I made my prints, so I activated and deactivated the timelapse just in case, but I still have the shift. 

You were testing with AMS, right? Could you test using only one color, so no changes in between?

BastR, 2024-09-20:
> > @XunZhangBambu As @jmilhau  says, the timelapse had never been activated when I made my prints, so I activated and deactivated the timelapse just in case, but I still have the shift.
> 
> You were testing with AMS, right? Could you test using only one color, so no changes in between?

Yes all my prints was with AMS, with only one color no problem.

jmilhau, 2024-09-20:
If I summarise:
- What works: Printing with 1 color
- What does not work: Printing with 2 colors or more with AMS/AMS lite with or without Timelapse activated

ziehmon, 2024-09-20:
@jmilhau @BastR  gotcha. That's the hint. I authored the change that probably caused this and I didn't had an AMS to test. But your description is enough to find it.

XunZhangBambu, 2024-09-20:
@ziehmon 
[image]
The bug occurs in this change. Although the retract and lift are executed before inserting the timelapse G-code, the actual lift will be performed during next travel generated by code, at which point the position may already be at the extreme edge of the printing area. Performing another spiral lift at this point may exceed the machine's printing range, leading to lost steps

BastR, 2024-09-20:
https://github.com/user-attachments/assets/68a75364-4976-4813-9588-51604d84e0bb

I deliberately shifted the X axis and you can see that during the last movement the head moves more towards 0.

ziehmon, 2024-09-20:
> @ziehmon [...]

I'm sorry, I don't agree with this analysis. There is plenty of room to execute the spiral Z hop. The problem is an inconsistency in the approached position (z=1.0000 vs z=1.2000) when there is an AMS in use .

 I am analyzing this, let's not jump to conclusions.

ziehmon, 2024-09-20:
[image]

ziehmon, 2024-09-20:
[image]

I found it. The position where the Z hop inserted via::
[image]

**is different, when AMS is used.** The head is at x=-48.2000 with AMS, and having a Z hop then indeed moves the head out of the bed's boundaries. 

I think we just have to change the gcode for filament_change @XunZhangBambu. The head needs to be at x=0.00000 after it, then everything works fine. That's the difference between the AMS gcode and the one without :)

XunZhangBambu, 2024-09-20:
@ziehmon  lets take the 3mf @jmilhau  provided to analysize.We add a comment where we expect the lift happen and set zhop to a special number 0.366
[image]

then you will find the lift dooes't immediately happen

[image]
it happens in the line 1440 , and the nozzle is out of the bed
[image]

[image]

ziehmon, 2024-09-20:
Yes you're right @XunZhangBambu. I agree and my analsys was only correct when no AMS is used. 

The fix is simple: in `filament_change_gcode`, the printer shall move to x=0.00000 at the end, exactly like it does in timelapse_gcode. This then gives enough room to perform any upfollowing moves:

```
G1 X0 Y{first_layer_center_no_wipe_tower[1]} F18000 ; move to safe pos
M400
```

ziehmon, 2024-09-20:
The screenshot you showed show exactly that this is because of the position the head is in when `filament_change_gcode` is finished! :)

XunZhangBambu, 2024-09-20:
@ziehmon What you mentioned could be a solution. However, in order to ensure absolute safety, we need to revert your submit first. After fixes and thorough testing, it will be reintegrated. Thank you for your understanding.

ziehmon, 2024-09-20:
Understood @XunZhangBambu. I'll create a new PR, including the change. Maybe you can start testing it soon. For a lot of people this fix massively improved print quality, but I understand that with AMS, this is a nasty bug.

ziehmon, 2024-09-20:
I am sorry for this mess. I don't own an AMS and I was unable to test this scenario. The fix for stringing is very nice but it's uncool that it broke people's printers. Sorry guys. At least we can do it better with the next PR :smoking:

SaltWei, 2024-09-20:
> I am sorry for this mess. I don't own an AMS and I was unable to test this scenario. The fix for stringing is very nice but it's uncool that it broke people's printers. Sorry guys. At least we can do it better with the next PR 🚬

This is not your fault. We always welcome PR from community, and our tester should tested more carefully and enough.
Sorry everyone, we will hot fix this issue ASAP.

jmilhau, 2024-09-20:
@ziehmon The fact that you are helping to solve the issue is way more important than the bug introduced. MANY THANKS for that. I would suggest Bambulab to send you an AMS for future testing ;)

ziehmon, 2024-09-20:
@lanewei120 thanks for your warm words. 

I created a new PR that handles this problem. There is a very extensive analysis inside: https://github.com/bambulab/BambuStudio/pull/4848. I also created a list of testcases we should print to make sure this is not going to happen again.

@XunZhangBambu would you be interested in reviewing and testing this, since you review the original PR?

ziehmon, 2024-09-20:
BTW, this was why the fix that caused this problem was introduced originally (printed without AMS....):

[image]

SaltWei, 2024-09-20:
> BTW, this was why the fix that caused this problem was introduced originally (printed without AMS....):

I will ask our market colleague whether it's possible to send you one ams lite(not promise). Will let you known if be approved.
:)

ziehmon, 2024-09-27:
> > BTW, this was why the fix that caused this problem was introduced originally (printed without AMS....):
> 
> 
> 
> I will ask our marketing colleague whether it's possible to send you one ams lite(not promise). Will let you known if be approved.
> 
> :)

That would be absolutely amazing! Even a discount code would be an amazing gesture, but it's understandable if it doesn't get approved 🙂

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.