bambulab/BambuStudio #4847: 打印机处于LAN模式下 MACOS版本触发休眠或锁屏 访问码不正确

bilibilifmk, 2024-09-20:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab github releases

### OS version

macos

### Printer

x1 A1

### How to reproduce

触发休眠或锁屏

### Actual results

[image]

### Expected results

正常连接

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FrancisNGG, 2024-09-26:
这个问题一直存在，挺烦人的。打印机的访问码又没有变更，客户端只要断开重连就要访问码，难道记住访问码很难实现吗？？？？？？？？？？？？

bilibilifmk, 2024-10-08:
稳定复现方法 开着软件 重启设备 必出现

walterwongbbl, 2024-10-08:
@FrancisNGG  @bilibilifmk 
This is a bug, we are trying to reproduce and resolve it.