bambulab/BambuStudio #4853: Support not generating interface on all tree

Mustangmvp, 2024-09-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

winfdows 11

### Additional system information

Windows 11 ‎(X64)‎ 23H2
AMD Ryzen 9 7950X3D Cadencé à 5.05 GHz
Micro-Star International Co., Ltd. MEG X670E ACE (MS-7D69) 1.0
AMD X670E
96 Go memory DDR5 6.6 GHz
NVIDIA GeForce RTX 4090

### Printer

Bambu lab A1

### How to reproduce

When I go into support, and want to put tree support, whether it's slim, strong, or organic, the software only puts interfaces on some and not on others, half of the support interfaces are missing.

### Actual results

[image]
[image]

### Expected results

support interfaces must appear

### Project file & Debug log uploads

[interface bug.zip](https://github.com/user-attachments/files/17078967/interface.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Mustangmvp, 2024-09-20:
[log.zip](https://github.com/user-attachments/files/17079027/log.zip)

tangketan, 2024-09-23:
It's the designed feature: not all the contacts need an interface. Too much interface will make the supports difficult to remove.

Mustangmvp, 2024-09-23:
I don't think so because in cura (lates verssion) i didnt have this problem the interface appear on all tree support branches