bambulab/BambuStudio #4854: VLC Complains about Timelapse Videos

youngbru, 2024-09-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Using VLC 3.0.21

### Printer

Bambu Labs A1 mini

### How to reproduce

Just open a timelapse video with VLC media player

### Actual results

VLC complains that the file is missing an index and requires an additional click in a dialog box t play the video.
[image]

### Expected results

It should just play like any other video file.

### Project file & Debug log uploads

Not related to any particular 3mf file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ados8, 2024-10-19:
I suspect it's to do with how the printer creates the metadata which VLC dislikes.

You can open in another video player and it's fine.

For those in favor of using VLC you can quickly fix with FFMPEG.

Command: ffmpeg -i {input} -vcodec copy {output}

Example, ffmpeg -i print.avi -vcodec copy print.mkv

You don't even need to change the container, AVI is fine, problem likely printer side.

To make the process less repetitive I've attached a batch script.

It checks all video files in the directory it is located for **.avi**

Any listed are converted without quality loss and appended with **_fixed_**

There's an option at the top which is default **false**, set to **true** if you want auto delete of the original video.

You will need to "install" FFMPEG.

1. Download FFMPEG: https://www.gyan.dev/ffmpeg/builds/ffmpeg-release-full.7z

2. Extract to your install programs install directory, rename FFMPEG

3. Copy path to bin folder i.e. C:\Program Files\FFMPEG\bin

4. Go to this PC -> properties -> advance system settings -> environment variables

5. Select path -> edit -> new

6. Paste the copied path and select ok, then again.

7. Paste the downloaded batch script to the media directory/video files.

Now when you want to fix the files run the script.

Script:
```
@echo off

setlocal enabledelayedexpansion

set "deleteOriginal=false"

if "%1"=="--delete" (

set "deleteOriginal=true"

)

for %%f in (*.avi) do (

echo %%~nf | findstr /b /c:"_fixed_" >nul

if errorlevel 1 (

set "input=%%f"

set "filename=%%~nf"

set "extension=%%~xf"

set "output=_fixed_!filename!!extension!"

echo Converting "!input!" to "!output!"

ffmpeg -i "!input!" -vcodec copy "!output!"

if "!deleteOriginal!"=="true" (

del "!input!"

echo Deleted original file "!input!"

)

)

)

endlocal
```

WayneBosworth, 2024-10-21:
This is 100% a bug in the printer firmware.  Please can this be looked at in an update for a proper fix.

gaolegao-lx, 2024-10-31:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

WayneBosworth, 2024-11-04:
In case this helps others, response from Bambu support...

Currently the video format is incompatible with VLC, we will try to solve it in the future version. It will take some time, there is no exact time for the moment.

For now, you need to click once more in the dialog box to play the video via VLC player. Otherwise, use a different media player to play the file such as windows media player.

ckizer, 2025-06-07:
Why hasn't this been fixed? I bought a brand new printer. It does this. It's BROKEN FIRMWARE software. I'm returning my printer. This feature is advertised as working, and it doesn't

ados8, 2025-06-08:
> Why hasn't this been fixed? I bought a brand new printer. It does this. It's BROKEN FIRMWARE software. I'm returning my printer. This feature is advertised as working, and it doesn't

I get that it's an annoying bug but it's really minor and there's easy workarounds.
That's like getting a great new car, it does the car functions really well, the seats are comfortable, it drives great, gets good millage, however the inbuilt maps is out of date so you flip out and try returning the car 😂.

ckizer, 2025-07-13:
> I get that it's an annoying bug but it's really minor and there's easy workarounds. That's like getting a great new car, it does the car functions really well, the seats are comfortable, it drives [...]

If my car advertised it has CarPlay but it didn't work when I got home, I would return the car for software updates. And if CarPlay still didn't work, I would return the car. I purchased the printer for the specific reason to make Timelapse movies, and it's clear from this thread that the bug isn't being worked on. It's being completely ignored for at least multiple months if not over a year.

WayneBosworth, 2025-11-12:
Over a year has gone by, this issue remains.  I've also found that the videos are not compatible with video sending via Whatsapp.  Disappointing.