bambulab/BambuStudio #4855: BambuStudio on Mac OSX Sequoia Crashing when sending to print

msardes, 2024-09-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OSX Sequoia

### Additional system information

  Model Name:	MacBook Air
  Model Identifier:	Mac15,12
  Model Number:	Z1BC00147LL/A
  Chip:	Apple M3
  Total Number of Cores:	8 (4 performance and 4 efficiency)
  Memory:	24 GB
  System Firmware Version:	11881.1.1
  OS Loader Version:	11881.1.1

### Printer

BambuLab A1 /AMS

### How to reproduce

Slice
Print Plate
Crashes

### Actual results

app crashes

### Expected results

app doesn't crash

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/17080119/Archive.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

skylerwshaw, 2024-10-13:
I am also experiencing this crash on version `1.9.7.52` with an X1C printer. MacOS 14.7. 2021 Apple MBP M1 Pro.

walterwongbbl, 2024-10-17:
@skylerwshaw 
We have identified a possible cause of this issue. Windows does not crash, but the underlying processing method of macOS is different, so we have optimized it. This optimization will be included in the next version.