bambulab/BambuStudio #4856: Download not working correctly for new version

sherrahs, 2024-09-20:
### Bambu Studio Version

1.9.7.5.0.

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I was prompted to download new version due to bugs with 19750,  I was prompted to download 10907 .52.  Which I have down 3 times now.  Every time it installs, its reinstalling the .50.   I cannot get the .52 to load.  It says it is installing it, but when it launches after install, its still the 0.50 version that was already on my computer.   I have shut it down and restarted it and restarted the download.  I am still getting the .50 version even though its says I am downloading .52.     

### Printer

A1 Mini

### How to reproduce

I am prompted upon opening bambu studio.  I click the link and am reidrected to the download site for .52.  I download and install, and watch it as it says it is installing .52   However, when it is done and I click launch.............it still launches the .50 version.   Even though I just downloaded .52.   

### Actual results

noted above

### Expected results

I would expect that I would have the latest version which ends in 52, since that is what I am being prompted to download.  I am not getting that after its done.   

### Project file & Debug log uploads

There is no project.  The project is Bambu studio prompting me to download a new version that is not working.   

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FiredEgg-bambu, 2024-09-21:
https://github.com/bambulab/BambuStudio/releases/download/v01.09.07.52/Bambu_Studio_win_public-v01.09.07.52-20240920143753.exe
take a try to download this

sherrahs, 2024-09-21:
Thank you.  The download you sent worked fine without issues and fast.  Its
installed.  Appreciate your help!!  Sherry Squires

ttt888zzz, 2024-09-23:
HI [sherrahs](https://github.com/sherrahs)：Please visit this website and download the latest version.https://github.com/bambulab/BambuStudio/releases，
[image]

sherrahs, 2024-09-25:
This link is not working.  This is a shot of the page it takes me too.
thanks.

[image: image.png]

ttt888zzz, 2024-09-26:
[image]
The website is OK？ https://github.com/bambulab/BambuStudio/releases