bambulab/BambuStudio #4857: Bambu Studio crashes after sending print job to cloud on Apple Mac Mini Sonoma 14.6.1

NEODNOC, 2024-09-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.6.1

### Additional system information

CPU Apple M2 Pro, 16GB RAM

### Printer

Bambu Lab P1S

### How to reproduce

Open an STL file, slice the part and upload to printer via Bambu Cloud. Crashes almost every time

### Actual results

Open an STL file, slice the part and upload to printer via Bambu Cloud. Crashes almost every time

### Expected results

Should not crash

### Project file & Debug log uploads

[BLAMSHubOutletToolV2.3mf.zip](https://github.com/user-attachments/files/17081164/BLAMSHubOutletToolV2.3mf.zip)
[Bambu Studio Crash Report Mac.zip](https://github.com/user-attachments/files/17081166/Bambu.Studio.Crash.Report.Mac.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

skylerwshaw, 2024-10-13:
I am also experiencing this crash on version `1.9.7.52` with an X1C printer. MacOS 14.7. 2021 Apple MBP M1 Pro.

walterwongbbl, 2024-10-17:
@skylerwshaw @NEODNOC 
We have identified a possible cause of this issue. Windows does not crash, but the underlying processing method of macOS is different, so we have optimized it. This optimization will be included in the next version.