bambulab/BambuStudio #4858: Drag and drop on home screen opens second instance of bambu studio

TennisRunner, 2024-09-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

Start bambu studio.
Click and drag a file into the window.

### Actual results

Opens a second instance of bambu studio

### Expected results

Should open the file in the current version.

You have to manually switch over to the prepare tab for the drag and drop to work even though bambu starts at the home screen.

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/17081244/Cube.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-09-23:
[image]
HI，[TennisRunner](https://github.com/TennisRunner)：Please refer to the options in the red box to modify your preferences

TennisRunner, 2024-09-23:
> HI，[TennisRunner](https://github.com/TennisRunner)：Please refer to the options in the red box to modify your preferences. [...]

That is a useful option to know, but if I drag a file into the current home screen of an instance of bambu studio. It should load the files on that instance instead of starting another instance.

[capture-9_23_2024 4-39 PM.webm](https://github.com/user-attachments/assets/85dc09ff-79dd-4029-a59e-7346b33145c6)